EFeru/hoverboard-firmware-hack-FOC #174: Hovercar variant

genetgenet, 2021-05-10:
Hi and thanx for this great project :)

I'm building an older kettcar with an hoverboard from 2014. It has the chip GD32F103RCT6. I unlocked the mcu but had a problem but after a while it worked somehow. I followed the hoverboard youtube clip and flashed the firmware with vscode, soldered the 3.3v wire for gas, connected thrustmaster pedals (pots) and tried it out. I have not soldered any pull down resistors "are they neseccary"? The motors spin up and then go slow no matter how much gas i give. I have tried to run on the floor but it just stutters forward on slow speed. I have tried to swap phase wires but both wheels only run at original wiring colors. Could the programming have gone wrong somehow? I'm lost here. I only have the st link no other debugging. I do have another older hoverboard with the stm32 chip but i would like to solve this first. When i swapped phase wires i have only swapped colors in one direction. I will post some pictures. Before the flash it worked like an ordinary hoverboard without problems.

genetgenet, 2021-05-10:
Uploading VID_20210510_212406.mp4…

EmanuelFeru, 2021-05-10:
The board you have might have slightly different pinouts. @Candas1 knows better. Have you tried with this macro set to 1?
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/0709e4cb62ef1d0bfedfc6bc3187614d5862acd6/Inc/config.h#L66

Candas1, 2021-05-10:
Hi,

The [alternate wiring](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility#pin-mapping) is for some boards with 6 wires on hall sensor cables.
Genetgenet, are you uploading a video ? it's not visible.
Have you tried the [autocalibration](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Input-Calibration) ?

genetgenet, 2021-05-10:
[image]

genetgenet, 2021-05-10:
Yes i have done autocalibration several times. The video did not upload for some reason. Here is the video link.

https://youtube.com/shorts/ClvuGA6BHeE?feature=share

Candas1, 2021-05-10:
Have you picked the right variant? Changed anything in config.h file?

genetgenet, 2021-05-11:
No i have not changed anything. I'm not familiar with vscode. I just followed instructions on the youtube hovercar video. Could something still gone wrong with the programming? As a said i had some problem unlocking the mcu, there were 3 boxes still checked that i could not uncheck as in the picture if it has something to do with this? (user configuration option byte) nrst_stop, nrst_stdby and wdg_sw.

genetgenet, 2021-05-11:
[image]

genetgenet, 2021-05-11:
I can understand some level of programming but i have not started to look into it yet.

Candas1, 2021-05-11:
No you don't need to uncheck those options it's ok.
If the firmware is in, the flashing went well.

Is anything happening when you press the brake pedal ? When you press it twice ?

Candas1, 2021-05-11:
Ok this is the way to activate reverse, only works at zero speed.
Just wanted to confirm as this option is only available in Hovercar variant

Candas1, 2021-05-11:
When wheels are not on the floor, braking can lead the wheels to oscillate forward/backwards, with a beep when going backwards, this is OK and will not happen on the floor.
But it still doesn't explain your problem.

Do you have a better picture of your mainboard without the cables in the middle ?

genetgenet, 2021-05-12:
https://youtu.be/NlG4vDOBUjg

genetgenet, 2021-05-12:
[image]
Gas pedal

genetgenet, 2021-05-12:
[image]
Brake pedal has an additional wire on the bottom

Candas1, 2021-05-12:
Your mainboard seems ok.
You need to check your wiring, you can use a multimeter

genetgenet, 2021-05-12:
Check all wiring? Can you be more specific? Wiring to pedals

genetgenet, 2021-05-13:
Hi again.

I connected 2 handlebar accelerator hall sensored but it did not work. Checked all cables found nothing wrong so i have redone the procedure from the beginning. First i used st32cubeprogrammer shown in another video to try unlocking mcu (even thou it should be unlocked) and i hot an error (the core is locked up) and failed to erase memory. The i unlocked through st link 004 version. Then i flashed with adc variant and it worked. One handlebar made both wheels go backwards and the other made both wheels go different direction. After this i have flashed hovercar and it works but it's a bit of a stuttery ride.

Another thing is with the sideboard. The program pins are in different location but i know + and -. Can i brake the st link if i put swdio and swdclk wrong?

Candas1, 2021-05-13:
No it shouldn't be a problem.

genetgenet, 2021-05-13:
I have the same issue as timtimmahh with the sideboard. The pinout is different with + and - on the end sides. I can't unlock the MCU with st link 004, with cube32 it seem to unlock but when i try to flash i get errors. I will post pictures.

genetgenet, 2021-05-13:
[image]
[image]
[image]

Candas1, 2021-05-13:
How are the option bytes looking like in st-utility? Can you take a screenshot?

Candas1, 2021-05-13:
Check WDG_SW there and try to flash again

genetgenet, 2021-05-13:
I have not tried to flash again after the comments above just yet. But i will :) 

Test drive :) 

https://youtube.com/shorts/hZqs0wvJkX8?feature=share

genetgenet, 2021-05-13:
Stutters at start but then it runs fine.

EmanuelFeru, 2021-05-13:
Did you try to swap motor phases? And do you have freewheel when pedal is 0?

genetgenet, 2021-05-13:
[image]
[image]

WDG_SW unchecked is successful but if i check WDG_SW i get the following error as in the pictures.

I have swapped the motor phases by clockwise (so to say) Yellow to Blue, Blue to green, Green to Yellow and so on. The wheels act strange in other phase swap. I have not cross swapped to other variations that don't make sense to me.

Candas1, 2021-05-13:
Can you use st-link utility software instead?

genetgenet, 2021-05-13:
[image]

genetgenet, 2021-05-13:
![ST link WDG_SW checked errorr 
[image]

Before flash

Candas1, 2021-05-13:
Can you try to flash now anyway?

genetgenet, 2021-05-13:
[image]
[image]

genetgenet, 2021-05-13:
The file is compiled with vscode and then i chose the firmware bin file in st link program and verify. Is this right ?

Candas1, 2021-05-13:
You can compile+flash in vscode directly

genetgenet, 2021-05-13:
[image]

SUCCESS. This is amazing. Thank you so much. I'm glad that this did not work out from the start. I've learned a lot and feel pretty confident of doing this procedure over again :) And thank you EmanuelFeru for all the help.

Candas1, 2021-05-13:
Yeah it's weird.
You enable an option, you get an error message, but then you can flash 🤔
Had the same with gd32 chip, need to document it.

genetgenet, 2021-05-13:
Yes it is. Thanks again. 

About the stuttering. The wheel is freewheeling but is making an annoying varience in sound like an irish sack pipe. i wrote before that i have swapped the motor phases by clockwise (so to say) Yellow to Blue, Blue to green, Green to Yellow and so on. The wheels act strange in other phase swap. I have not cross swapped to other variations that don't make sense to me.

EmanuelFeru, 2021-05-13:
In Torque mode is normal to have some noise, because the current measurement is not perfectly clean and it affects the command.
But the stutter is strange, it should not happen normally. Have you tried other mode, like Voltage to see if it still happens?

genetgenet, 2021-05-14:
I will test voltage mode. I will have to read up on how to do it.

closed 2022-01-02 by Candas1