EFeru/hoverboard-firmware-hack-FOC #176: Problem with left wheel

mizerakle, 2021-05-11:
Hello! We have problem with second wheel. we make a rc cart(we used pwm). wheel steering when the wheels are not touching the ground is excellent. when moving on the ground, if the cart collides with an obstacle, one wheel stops and the second continues to rotate, sometimes turning off the radio control.  when the wheel is lifted off the surface, it stops. When changing motors in places, the defect moved from one wheel to another. the error always occurs on the motor that is connected to left hall cable. the order of colors when connecting the motor and hall sensors is kept original (from the factory). We ised variant pwm and change variables CTRL_TYP_SEL FOC_CTRL , CTRL_MOD_REQ TRQ_MODE, M_MOT_MAX 100. the rest is default.

mizerakle, 2021-05-13:
tested on two different boards, first noverboard on gd32, second on stm32, the error is the same.

EmanuelFeru, 2021-05-13:
Did you check with another battery? Maybe when you hit an obstacle, there is but Voltage drop or current and strange things are happening.

mizerakle, 2021-05-13:
Yes, I change 3 different batteries, error is the same

mizerakle, 2021-05-13:
I used another (3)board(gd32),everything worked well on it.  But this mother board died before flashing(the power wire fell off when I flashing board.

EmanuelFeru, 2021-05-13:
So you had one board that worked ok? 
The other 2 boards are the standard ones? I mean do they look similar as the one in the Readme picture?

vamfun, 2021-05-14:
Check out issue #129 re failure to connect to St-link.... sounded like your board died during download.....   if you corrupted the firmware then there is no SWDIO and SWCLK  Pins activated to take the download so you need to use the procedures discussed in #129 to get your board going again unless of course the board is fried.     I almost threw my board away till @Candas saved me😎

http://vamfun.wordpress.com
Robot musing blog
iPhone

closed 2021-07-24 by EmanuelFeru