EFeru/hoverboard-firmware-hack-FOC #177: Q: Wich is the best control method from Arduino? ... and why ?

mrlunk, 2021-05-12:
Hi again all :)

As mentioned in the title:
Q:  Wich is the best control method from Arduino? ... and why ?

- PWM
- PPM 
- UART

GreetZ,
Peter Lunk

Candas1, 2021-05-12:
Hi,

Please explore the wiki.
You will get to know more about the firmware and find the answer to your question

Candas1, 2021-05-12:
And the main page also 😜

mrlunk, 2021-05-12:
Well I can send all three from an arduino with some effort...
Why is UART advised ?

Candas1, 2021-05-13:
Have you read this ?
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Variant-USART

You see that the transmission errors are rejected, you can get feedback data, ....

mrlunk, 2021-05-13:
Ah ok, ...
So sending PWM or PWM would not give any feedback data ?

(I can program failsafes for PWM RC data transmission fails too)

PL.

closed 2021-05-13 by mrlunk