EFeru/hoverboard-firmware-hack-FOC #179: Iq variation

goosst, 2021-05-24:
### Your System
- VARIANT: VARIANT_USART
- CONTROL TYPE: FOC_CTRL
- CONTROL MODE: VLT_MODE
- What is your application?
-- one day a windturbine

**Before asking a question** see if it is already answered in:
* [Wiki pages](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki)

**Indicate what you have tried:**

Again, thanks for the great firmware. Might not be an issue but maybe more a question.

I was free spinning the (left)motor with some blades attached to it and monitoring the Iq current.
- overwrote this message in main.c: `Feedback.cmd2 = (int16_t)rtY_Left.iq; //overwrite`, to have some feedback about the torque
- voltage commands were give over uart3 (so no noise by ADC etc. expected)

Printed the Iq every 500ms, do you consider it expected to have this amount of variation on iq?

p.s.: are there parameters like torque constant / back-emf constant known about these hoverboard motor? 
I'm just wondering how accurate the dc_link_current is to do power calculations or if a Iq based method would be better.

```
  timestamp:1642047  left_dc_current[cA]:42  batt_vlt:4081  spd_act_lft[rpm]:80  iq:779  temper:261
  timestamp:1642551  left_dc_current[cA]:36  batt_vlt:4081  spd_act_lft[rpm]:83  iq:818  temper:261
  timestamp:1643055  left_dc_current[cA]:38  batt_vlt:4081  spd_act_lft[rpm]:80  iq:749  temper:261
  timestamp:1643549  left_dc_current[cA]:40  batt_vlt:4081  spd_act_lft[rpm]:80  iq:770  temper:261
  timestamp:1644053  left_dc_current[cA]:42  batt_vlt:4079  spd_act_lft[rpm]:80  iq:610  temper:261
  timestamp:1644547  left_dc_current[cA]:20  batt_vlt:4079  spd_act_lft[rpm]:81  iq:700  temper:261
  timestamp:1645051  left_dc_current[cA]:40  batt_vlt:4079  spd_act_lft[rpm]:82  iq:1013  temper:261
  timestamp:1645555  left_dc_current[cA]:38  batt_vlt:4079  spd_act_lft[rpm]:81  iq:762  temper:261
  timestamp:1646049  left_dc_current[cA]:22  batt_vlt:4079  spd_act_lft[rpm]:82  iq:769  temper:261
  timestamp:1646553  left_dc_current[cA]:20  batt_vlt:4079  spd_act_lft[rpm]:81  iq:746  temper:261
  timestamp:1647047  left_dc_current[cA]:38  batt_vlt:4079  spd_act_lft[rpm]:80  iq:690  temper:259
```

EmanuelFeru, 2021-05-25:
Hi,

I am not completely sure about the accuracy `I_DC_Link` vs. `iq`. However, if I remember corectly the `I_DC_Link` is more steady and less prone to spikes, so it might be more suitable for your calculation. But in essence you can use both. If you have some means to check (e.g., external current measurement) to see which one gives better results, that could help in the selection.

If you use the `rtY_Left.iq` don't forget to convert from fixed-point `fixdt(1,16,4)` back to normal values in A, i.e.:
```
Feedback.cmd2 = (int16_t)(rtY_Left.iq >> 4);
```
or if you want don't want to loose precision, then you have to use `float` and division:
```
val_float = (float)(rtY_Left.iq / 16);    // 16 = 2^4
```

goosst, 2021-05-27:
thanks for the conversion factors :) 

Seems like the Iq based power is a bit more stable compared to dc-link (had to guess the Kt factor however).
[image]

closed 2021-07-24 by EmanuelFeru