EFeru/hoverboard-firmware-hack-FOC #180: out of control when pwm wire lost

jpfree, 2021-05-26:
thank you very much for your work and contribution。

I encountered a very terrible thing, when the PWM line of the receiver is broken or the receiver fails, the wheel will turn uncontrollably. It's the same with raspberry pi When raspberry pi restarts, 

i must connected the pwm line to the raspberry first,then  power on the raspberry，and  power on the board,it will work fine。

how can I add an initial PWM signal to the firmware? Similar out of control protection function, thank you

EmanuelFeru, 2021-05-26:
It is most probably moving because the line is floating and it's detecting false pwm pulses. 
You can try to add 100k pull down resistors, so the timeout functionality will trigger and as a result the motors will stop.

Candas1, 2021-05-26:
Hi

Using pulldown resistors was also suggested in the telegram group.
Also I was asking why using PWM variant from a raspberry pi when you can use [USART](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Variant-USART) Variant ?

closed 2021-07-24 by EmanuelFeru