EFeru/hoverboard-firmware-hack-FOC #181: Possible values of z_errCode

swhof123, 2021-05-28:
Ran my first tests yesterday with my remote controlled lawn mower. 🙂

One issue I would like to solve, is around „motor blocked“:

When driving into grass that is too high, the wheels get blocked and the mower shuts down. This is totally fine to protect the hardware. What is quite annoying is that I have to switch the board off and on again to continue. 
My plan would be that the boards stops at „motor blocked“, as defined right now. But that I can still drive it in the opposite direction. 
So, I would extend the „else if“ in main.c:491 (section  ## BEEP AND EMERGENCY POWEROFF ## ). But I have not information on the values, that z_errCode can have. 
Could you please provide a list of possible values?
Also, are the error codes distinct enough, so that I can narrow it down sufficiently to the „motor blocked“ error?

- VARIANT: UART
- CONTROL TYPE: FOC_CTRL
- CONTROL MODE: SPD_MODE

EmanuelFeru, 2021-05-28:
The possibilites of `z_errCode` ar explained in this repo:
https://github.com/EmanuelFeru/bldc-motor-control-FOC#diagnostics

The error is per bit. So, it is possible to have 2 or more errors in the same time. If you are interested for a specifict error then you have to do some easy bit calculation, e.g., `mot_err = z_errCode & 0x04`.

For „motor blocked“ error is the Error 4 you need to check. But the thing is that if you get the "Motor blocked" errors it can only be reset by a **board restart**. So, my advice is better to avoid the error by setting this threshold higher (double the value below) and manage the behavior yourself (by checking the reading the command value sent to the motors and do some actions by sending a differen Speed target, for example negative to go reverse, to meet your needs.
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/eb20cc0acad0da91ead7d207fa6a83f1778f03e1/Src/BLDC_controller_data.c#L271-L274

swhof123, 2021-06-12:
Thanks Emanuel for the explanation. 
I will do as you suggested and add a custom threshold with some custom action to completely avoid the Error 4. 

Related to that: When "Motor blocked" Error is raised, the board also disables stand-still-hold... which might lead to the board rolling downhill unexpectedly. Another reason to really trying to avoid this error.

closed 2021-06-12 by swhof123