EFeru/hoverboard-firmware-hack-FOC #182: Nothing expect constant slow deep beep

mind999hacker, 2021-06-01:
### Your System
- VARIANT: ADC
- CONTROL TYPE: 2 potis
- CONTROL MODE: 3
- What is your application?
    - Hovercar, Skateboard, RC platform, etc. just testing 
**Before asking a question** see if it is already answered in:
* [Wiki pages](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki)

**Indicate what you have tried:**
If applicable, indicate what you tried that _doesn't_ work:
I tried changing input types calibrating input changing potis nothing works. It just has a slowish beep continuous and nothing responds. Slow beeping means low battery according to the diag but changing calibration values in config.h does nothing. I was able to use https://github.com/p-h-a-i-l/hoverboard-firmware-hack this hover.hex file in the build folder using the potis and it was able to work. I am compiling and uploading using platformio. Any idea what might be the issue?

mind999hacker, 2021-06-01:
Ok update after further look it looks like there is an input limit without enough hall inputs it wouldn't work. After plugging in both motors it seems the issue is gone. It would be nice to be able to test with single motor

Candas1, 2021-06-02:
Check config.h, you deactivate a motor if needed

closed 2021-06-23 by mind999hacker