EFeru/hoverboard-firmware-hack-FOC #183: Use as e-bike

serbasan, 2021-06-14:
Hello. I've been with a bike for a few years like "TheHackLife" in this video [https://www.youtube.com/watch?v=1PNCsTchDvg](url) with the original firmware [.](https://github.com/NiklasFauth/hoverboard-firmware-hack) It was pretty easy to operate on the throttle.
Now I have seen this new project and I have some queries:
1) What do I have to change in the ADC variant for use as an e-bike?
2) What parameters do I have to change to regulate the maximum speed?
3) If I stop accelerating the wheel is free?
4) What do I have to change to manage battery consumption?
5) Is my wiring configuration sufficient? Can I install this firmware with the wiring I already have? (seen in the video)

Thank you very much and waiting for answers to start testing this on my e-bike

EmanuelFeru, 2021-06-14:
Nice build! Yes it should work with your writing. Have a look at the wiki pages, all your questions are answered there.

serbasan, 2021-06-16:
I am amazed at how well this works. Very nice to ride the bike with torque. The behavior is very smooth at the start and when it reaches speed it starts to consume more and reaches supersonic speeds (40km with 20 ", we are crazy haha. FIELD_WEAK_ENA is cool)
One question, what parameters would allow to give more TORQUE when starting and less torque as speed is reached? I'd like to adjust it a bit.
Thanks for the job. Very thankful.

creathinq, 2021-06-17:
hi. I have exactly same issue, but with hovercar variant - poor torque at very low speeds (so almost none hill climbing ability) but as it picks up some speed, then serious torque kicks in and I have trouble hanging on to my e-kickscooter =). Im not hitting current limit there, I measured it. I think it does not matter if I use hovercar or ADC variant, right? I use torque mode of course.  Maybe there is some torque curve adjustment, but I can't find it.

EmanuelFeru, 2021-06-18:
There is RATE and FILTER in config.h to adjust the response to throttle during dynamic input changes. However if you want to adjust the torque curve (it should be more or less linear now) then you have to do it on the source side , for example in the remote if you use one. Or in the firmware to add a map between the Raw ADC reading and input1, input2. This map is not implemented, so it is 1 to 1 at the moment.

creathinq, 2021-06-18:
I`m using ebike thumb throttle (hall type), there should be no problem with that, many are using this. maybe I need to use both motors instead of only one.  I`m curious why such poor torque from standstill, it draws about 8A during initial acceleration and only later with some speed it reaches current limit (set at 12A now).  I will try to play with RATE and FILTER then

benjaf, 2021-06-18:
Actually, I have been having the same issue. In the near-zero rpm range (where commutation is used until FOC kicks in?) there is significantly less torque. This applies to both Speed and Torque modes. I have not have the chance to use a recent build, but it seems the behavior has not changed. Maybe try the commutation control mode in stead of FOC and see if this changes anything?

serbasan, 2021-06-18:
Tried with Hall and Probe and the torque changes very little at the beginning, it is very little noticeable.
I love the FOC mode because it frees the wheel and is very enjoyable on the bike.
For now I will continue with this setup even if I have to do a little more effort in the hills.
I will be pending the development of this firmware in case there are improvements.
Thanks to EmanuelFeru.

creathinq, 2021-06-20:
Tested my scooter a bit. It looks like torque is proportional to actual speed too (not only input). If I roll dowhill (0 input) a then punch it, scooter accelerates swiftly. but going uphill is impossible. full throttle and as I run out of speed and inertia, it stops. motor current limit is set to 12A and I only measured that on level road when going quite fast. accelerating from standstill takes 8A max.

EmanuelFeru, 2021-06-20:
That can be. In the firmware there is a constant factor to convert from ADC reading to current, but that conversion is most probably not linear. So, it needs a nonlinear mapping to get current values in line with real measurements. However, on the other hand, for such application we don't need such accuracies. 
I my suggestion is to increase the current limit if you observed that at standstill it doesn't do the limit.

creathinq, 2021-06-20:
> My suggestion is to increase the current limit if you observed that at standstill it doesn't do the limit.

yes I could, but then I might fry old battery after picking up some speed :).  I wonder if single motor is just not enough here. 
I`m building a gokart for my kids with 2 motors, will see how that one will go.

benjaf, 2021-06-21:
In my application it didn't seem like increasing the current limit past a certain point made any significant change to standstill torque. The gokart I'm building can barely get going on grass (with an adult on it), but once moving it accelerates quite quickly.

creathinq, 2021-06-21:
> In my application it didn't seem like increasing the current limit past a certain point made any significant change to standstill torque. The gokart I'm building can barely get going on grass (with an adult on it), but once moving it accelerates quite quickly.

That is bad news. I only need low top speed (for kids), but good starting torque. maybe FOC-TRQ mode is not suitable here at all? original hoverboard has tons of standstill torque, would climb a wall.

benjaf, 2021-06-21:
> That is bad news. I only need low top speed (for kids), but good starting torque. maybe FOC-TRQ mode is not suitable here at all? original hoverboard has tons of standstill torque, would climb a wall.

I have also used the NiklasFauth firmware, and it provided a much higher starting torque with a similar configuration. I don´t know where the limitation is with this version, but it definitely behaves differently at low speed.

juodumas, 2021-06-21:
I am using sinusoidal with this firmware and standstill torque is great. Didn't try FOC TRQ.

serbasan, 2021-06-21:
> Estoy usando sinusoidal con este firmware y el par de torsión en reposo es excelente. No probé FOC TRQ.

For the bike we want the wheel to be free when releasing the accelerator. I think the only mode that the motor is free is with torque, so we did not try other modes. All the best.

creathinq, 2021-06-22:
I tried SIN_CTRL in VLT_MODE. huge torque from standstill. I still did not make it to the very top of my testing dirt road (8m long, about 2m rise, that sums up to 25% or 14° slope), but I did get further than on TRQ mode. I might be on 15A I_MOT_MAX  limit, or the battery is done. I'm probably asking too much of single motor and original used battery, with my 100kg (+scooter). anyway, without freewheeling, it is unusable in scooter/ebike.

juodumas, 2021-06-22:
Yes, you will need 2 wheels for such load.

benjaf, 2021-06-22:
You will need a better battery as well if adding another motor, even one motor can easily pull more current than one of the stock batteries can supply. 
It seems I will have to fix the gokart to do some further testing of the different modes!

vamfun, 2021-06-22:
I wanted to get an idea of the steady state speed that can be maintained from the power physics.  I was curious since I’ve never built one.
V = P*100/(W*%*g).   Where V is in m/s , P is in Watts, W is weight in kg , (% is percent slope, g is gravity = 9.8 m/s/s, 
So V = 10.2 *P/(W*%).  
Example.  W= 180 kg   (100kg ebike plus 80 kg rider ) 
                    P =  250 watt motor typical hoverboard 6.5 in motor @15 amps 
                    %= 20
                 ==>   V = 0.71 m/s.  1 motor, 1.42m/s with two motors.  
The efficiency of the ebike in transferring the power to the wheels ..and the larger wheels used for off-road hoverboards are some big variables… but are you guys seeing speeds on this order?       I think the larger 8.5” or  10” hoverboard  wheels don’t give much more power …just higher speed and ground clearance based on robodurans videos, right?

http://vamfun.wordpress.com
Robot musing blog
iPhone

benjaf, 2021-06-22:
Well, regarding the 8.5" and 10" wheels, I have yet to see one where the stator isn't virtually the same as the 6.5". All that is changed is the outer rim of the rotor, allowing for different tyre options.
And I don't know about the efficiency of these motors, but driving one at ~40V 15A would make the P value 600 rather than 250,  at least if efficiency is assumed to be 100% and any resistance is ignored, correct?

vamfun, 2021-06-22:
Yes I agree the stator size is about the same.   When I spoke of 15 amps..that is the maximum current.   Current at peak power is typically 50% max current.    Most bldc motors are about 60% efficient at peak power which is how most manufacturers quote their power.
The efficiency takes into account the I2R losses and bearing friction losses etc.
Maximum BLDC efficiency  can be around 85% compared to brushed motors at 75%.

So one would guess the peak power at   40v*7.5amps*.6 = 180 watts.
Power at peak efficiency would be about .5*peak power = 90 watts.

To get 250 watts we would need a maximum current of 20.8 amps.

I don’t have a hoverboard motor curve but here is a typical bldc characteristic 

http://vamfun.wordpress.com
Robot musing blog
iPhone

Candas1, 2022-04-20:
In voltage mode? Not torque mode?

JSphere22, 2022-04-21:
"In voltage mode? Not torque mode?" 
Torque mode
I changed #define STEER_COEFFICIENT       0   // -0.2f (changed from 62259 to 0 to try to disable reverse)

Candas1, 2022-04-21:
If you use the auto-calibration and move only the joystick forward and in one direction it should work.

Otherwise you could try [SKATEBOARD](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-SKATEBOARD) variant and replace CONTROL_PWM_RIGHT by CONTROL_NUNCHUK [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/bd4a34dabf655ba8a9a48e9974ebb0b619796fde/Inc/config.h#L583)

It would allow you to use throttle and brake on the same axis.

JSphere22, 2022-04-21:
OMG I am so pissed. I tried to disable reverse by changing  NUNCHUK 
"#define AUX_INPUT1            3, -1000, 0, 1000, 0 "  to
"#define AUX_INPUT1            3, 0, 0, 1000, 0"  
Bricked the board. Tried to flash another now just getting 4 beeps. Everything was working fine.

Candas1, 2022-04-21:
You need to keep the button pressed when flashing. If it's not working you probably killed a tip127

JSphere22, 2022-04-21:
I have  some tip127 I'll change them out tomorrow. Thank You

JSphere22, 2022-04-22:
tip127's did not repair no power boards. I am going to order another to see if I can get it and use either skateboard or just calibration. Thank you for your help!

PuceBaboon, 2022-04-23:
JSphere22,

   Just a quick sanity check (I'm trying to understand your situation).  Are you getting a -repeated-  4-beep sequence?  That would indicate that the board isn't seeing the nunchuk controller (I don't have a nunchuk, so I'm just basing this off my own experience of using UART/Serial).  Make sure you're getting 3v3 and a good ground to the nunchuk and that your SDA/SCL cables aren't switched round (it won't harm either the nunchuk or the motherboard if they are swapped, anyway).

   Good luck!

    -John-

JSphere22, 2022-04-24:
I'm thinking that you maybe right on the 4 beeps. However I flashed another that is working with my backup nunchuk not the same nunchuk (original beeped 4 times). I tried the calibration to disable reverse, but it disabled input. Worked again after full calibration. 
My next step would be to use the skateboard variant, with nunchuk input. My question is would it still freewheel with the variant?

JSphere22, 2022-04-24:
On my other board that beep 4 times it locks up during calibration. I'm thinking I need to wipe flash memory.
How do you change this key?        
[#define](https://www.youtube.com/hashtag/define) FLASH_WRITE_KEY      0x1002  // Flash memory writing key. Change this key to ignore the input calibrations from the flash memory and use the ones in config.h

Candas1, 2022-04-25:
1. You should create a new issue if you have questions, a lot of people are getting notified about your questions now
2. If you get 4 beeps, it has nothing to do with the calibration