EFeru/hoverboard-firmware-hack-FOC #184: e-mower

s56jsc, 2021-06-19:
Dear hover-lover 
After long, long time I do finish my little project. This is RC mower.
I use old mower, 2. hover-borads (10" and 8.5" wheel) old drone command, arduino nano for communication and mega for hover-board communication.
It works on table :) In real word not. One controller if wheel is not rotate "going to tilt". This is strange, because I use same FW for two different boards. How to solve this problem? Also I note, that wheel have to low torque. I know that it will be better idea to use both 8.5" wheel. But I think it is not enough for climb on my 45% slope. I have filling that hover board was stronger?!?!
Here is my first test https://youtu.be/hxyF0Ze5qa0
And some pictures.
[image]
[image]
[image]
[image]

s56jsc, 2021-06-23:
Do someone have idea, why motor's have so low torque? Is this normal? I have filling, that hoverboard is stronger, and can push this mower without problem...

EmanuelFeru, 2021-06-23:
Normally the torque of 2 motors is enough to move a person. So 4 motors should do double. Basically, it should normally climb the wall with 4 motors.
Did you check in the debug the MIN and MAX of your input? I am afraid you are not using the full range... probably you didn't do the calibration right, I am not sure. You need to check.
I don't know what mode are you using, but probably for your application best is SPEED_MODE because it will use full torque even at low speed to maintain its speed.

s56jsc, 2021-06-23:
Dear Emanuel
Yes, I'm using a SPEED mode, because it is better to break wheal when going down-hill.
Calibration? Hmmm I think I miss this part. I didn't know that it need to be done. I only build your file and upload it to controller. How do do it?

EmanuelFeru, 2021-06-23:
Check the wiki pages

vamfun, 2021-06-23:
Lawn mower Torque calc 
Max torque for a hoverboard setup is around 12.5 nm.  (110.6 in lb) at zero speed and 15amps.
Max force 8.5 in wheel = 110.6/4.25 = 26 Lb
Max force 10.  In wheel = 110.6/ 5 = 22.1 
With two 8.5 in and two 10 in wheels the
total max pushing force = 48.1*2 = 96.2 1b

If mowing at 80% torque to have some speed 
Then total max force = 77lb

45 deg slope will cause a gravity drag of W*sin(45) 
where W is the mower weight.
Net pushing force = 77-W*.7
Weight lb
Net force #
50
42
75
24.5
100
7
125
-10.5
150 
-28
You might have a chance with a mower weight between 75 and 100 #. 
If you switched to 6.5 in wheels the max pushing force @ 80% torque = 109 #  or 32 # extra pushing force.    Then your mower can weigh more.

With your current configuration you might get away with a mower that weighs between 75 and 100 # … but with smaller wheels your mower could weigh between 125 and 140? .  It all depends on the grass drag force.   

This is the best case scenario..   you probably can’t run continuously at 80% torque because the current is around 12 amps.     Continuous torque is usually 30% to 40% max for most motors but haven’t tested this myself.  So currents around 5 amps should be sustainable depending on cooling.   If we’ll cooled maybe 8 amps?  Emanuel has better data on this.   You can repeat the calculation for lower currents.
Good luck

http://vamfun.wordpress.com
Robot musing blog
iPhone

s56jsc, 2021-06-24:
Thanks guys
I will try it during weekend!

s56jsc, 2021-06-26:
Hi
Today I read wiki page,but not help me too much. I also install FPViewer, but I cannot find out how to use it. I'm sorry I'm not engineer :(
@vamfun Thanks for calculation. You are not like mi. First try, then repair :) Total weigh is 43kg (95lb). So in theory should work. Problem is, that terrain is not flat, so at some moment only 2. wheel will touch ground. Also with smaller wheel I loose some grip. 8.5" I think it is minimum. Also 45 deg slope is not all the time. 
Main problem here is that one only board seen's to have HW current protection. When wheel cannot turn (whistling sounds from the wheel)  it go to error board start to beep and release wheels. Other board with same FW doesn't do this. I also revers front and real motor. Same problem. So problem is on board not motor.
Other is, that I think that controller don't give enough power to motor. My power clamp show 18A@controller at battery at maximum. (motor blocked) 
I have I_MOT_MAX set to 15 and I_DC_MAX set to 17 (default settings) 
[image]
I have no idea what to do now.

EmanuelFeru, 2021-06-26:
Some sort of suspension system will help a lot. I heard some people having the same issue, wheels not touching the ground ask the time. For your application i think is important to have all wheels on the ground. Maybe you can find some flexibility, maybe one axle to pivot in the center relative to the other.

You said one board doesn't give full power. Did you have another board to test with or have you performed calibration and you forgot, values are stored in memory and not erased even after reflash. Do a full chip erase just to be sure.

s56jsc, 2021-06-26:
True, this is not ideal system, but for test it should be OK. With pivot axle can be save this problem.

My system have 2. boards. One is work with 18A, but it at lest work. Other make me more problems. It is working until one of wheel is not stooped. In this case board cut power for both wheels and start beep. On board is STM103 MPU. I try to full erase chip, but Ii got an error 
[image]

If I select erase  sectors and select all it will display like OK, but program remind. If I select Option Bytes.. and enable, then disable protection will erase chip, but after programming have same issue
[image]
On other (working) board is totally same program. I do check all MOSFET and looking good.

s56jsc, 2021-06-26:
Also video what is happens when wheel is blocked
https://youtu.be/VsTJo45Ke8A

EmanuelFeru, 2021-06-26:
What is that noise before the beep? Can be that you have to swap the motors phases? Because of mismatch.

s56jsc, 2021-06-26:
This is sound only when wheel are blocked. On other case it spin normal. 
I was think too, that it is phase swab, but I try all all combination.

EmanuelFeru, 2021-06-26:
I understand but is not a normal sound. It sounds like the current chopping is kicking in. If in reality is doing much less current I think there is something wrong with the board, because it's measuring more current than normal.

s56jsc, 2021-06-26:
I think too. I will buy new board and try again. Strange is that hoverboard is working normal.

thanks for all help for now

vamfun, 2021-06-27:
The weight of your mower puts you at very little margin at 45 degs my guess the physics are against you particularly if one wheel is not contacting ground.
   A big unknown is the drag of the grass. Maybe characterize this by finding uncut grass patch on flat ground and drag your mower (zero current to wheels) through grass at the desired speed of cutting .    Use some elastic cord and measure the deflection while pulling and then duplicate the deflection with hanging weights.  Better yet use a luggage electronic weight scale if you have one.
Next run your mower up a ramp and find the maximum angle you can sustain.  Make sure the wheels aren’t slipping.   These two data points will tell you if you need a redesign.  

It would certainly help to have same size wheels and couple wheels together on each side.
I would not rule out the 6.5 in wheels as a simpler solution.
Good luck again.

http://vamfun.wordpress.com
Robot musing blog
iPhone

s56jsc, 2021-06-28:
Thanks for this
45% is maximum, but normally is 25-30% Also this can be cut in parallel. Now first I must found new hoverboard with 8.5" wheel and change driver board. Then next step.

thanks for all help!

s56jsc, 2021-06-28:
I found XPLORE XP9718 hover board with 800W motors. I hope it can be upload EmanualFeru's FW

vamfun, 2021-06-29:
Hi..sorry, I misread your 45% to be 45deg slope.
I know the weight of your mower now and can redo the calculations as a function of % slope.  I owe you this😎
% slope will cause a gravity drag of W*%/100
where W is the mower weight.
Net pushing force = Max force#*%max force/100-W*%/100.   
With a 100lb mower net force and and 100lb max pushing force this simplifies to

net force #= (%max force- %slope)

             Net pushing force vs %slope
Slope %
Net ***@***.***% max force or  12 amps
Net force @50% max force  or 7.5 amps 
15%
65#
35#
20
60
30
25
55
25
30 
50
20
35
45
15
40
40
10
45
35
5

So things are looking rosier… with four wheels on the grass.. 
but with three wheels you can still make 45% slope @80% power but only 15% slope @50% power.  With only two wheels touching grass, ie going over a rock with no suspension…then you can lower the net forces in table by approx 50#. … 
 Chained together wheels can help a lot in this situation.   

http://vamfun.wordpress.com
Robot musing blog
iPhone

EmanuelFeru, 2021-06-29:
Another idea is to move front wheels to the back, basically all traction wheels close to each other to maximize the contact with the ground. And in the front put some dummy caster wheels. Probably it will steer easier too, because rotation point is closer to the wheels.

s56jsc, 2021-06-30:
True, but If I wont to keep low gravity weigh it must be done like this. I waiting now for 400W wheels and driver. Hope your FW will support it :)

s56jsc, 2021-07-11:
I buy new hoverboard with 400W motors. CPU board look like original Manuel's. I upload FW and work. This is great!
I mount it on my mower (with old 10" wheels) but the same. Go in error if I hold wheel. I mount original 8.5" wheels, but error remain. I double check all cables and connections and they are OK.
I really have no idea what to do now. I will understand it it this normal. But front wheels with same FW (but different CPU board) work OK.
Sad, verry sad :( :( :(

EmanuelFeru, 2021-07-11:
Could be that one motor has some defect maybe? A winding, or hall sensor? I don't remember, did you try to run the working motors with this new board?

s56jsc, 2021-07-12:
Problem here is, because there is new CPU board and new motors. I try before hoverboard and it was working.

If I connect motors to other board it works. If I connect motors from working CPU board to new board it is not work.
So problem here is 100% in new CPU board.
I try with osciloscope to look hall sensor signal, and they are OK.
motor phases are OK
In fact everything work fine. Spin forward, backward, ... Only when one of wheel is blocked it is going to error.

Why one board is work well other not. Here is good question.... I guess some protection, but how to disable it ?

s56jsc, 2021-07-13:
I put old CPU board on my table and now I'm trying to change code (by guessing I'm not programmer) and hope to get work. Hope I will have some luck.....
This is video from original what is happens when I block wheel and try to spin. This are 2. phases of motor.
https://youtu.be/lsyRirWutts

s56jsc, 2021-07-20:
After a lot of trying and testing I got some result, but I'm not satisfied yet...
If I use I_MOT_MAX 12 or less and I_DC_MAX 25 or more then error it is not happens any more. But I need full power for my mower.
By logic (I only use left wheel right is disabled) If I change input of DC bus current (PC0 ->PC1) and phase current (PA0, PC3 -> PC4,PC5) it should not measure current and not give me error, but the same...

I'm totally stuck here :(

s56jsc, 2021-07-23:
giving up. Cannot find solution.....

closed 2021-07-23 by s56jsc