EFeru/hoverboard-firmware-hack-FOC #185: Prediction or measurement of wheel position

ducktrA, 2021-06-21:
i am would like to access the best position estimation in the code. i see that "a_mechAngle" is never assigned apart from 0.
"a_elecAngle" reveals access to the 6 hall states. is there a better place in the FOC code where this information is there or is hooking in the interrupt routine for encoder state reading the best place to sum up wheel counts to build an odom source.

Thank you

vamfun, 2021-06-22:
here is some motor position code that I am using successfully 

``void Motor_Pos(void){
	//This routine computes the cumulative rotation of the wheels measured in electrical degrees
	//It takes about 5us to run and should be called by bldc.c ...I put it at the end.
	//inputs are either analog encoder mechanical degrees read on Left Sensor ADC tx and rx inputs respectively
	//or from internal hall sensor output variables motAngleLeft and motAngleRight found in bldc.c
	//Selection is set by #define ANALOG_ENCODER_ADC set in config.h
	//outputs to global variables MotorPosL and MotorPosR
	//note...when looking at the wheel exterior, counter clockwise rotation is positive.
	//author:vamfun@yahoo.com 
uint16_t AngleLeft = 0;// cyclic angle variables scaled in motor elec degrees
uint16_t AngleRight = 0;		
static uint16_t AngleLeftLast = 0;//
static uint16_t AngleRightLast = 0;	
static int cycleDegsL = 0 ;  //wheel rotation roll over count in elec deg
static int cycleDegsR = 0 ;  //wheel rotation roll over count in elec deg	
int16_t diffL = 0;
int16_t diffR = 0;
static int cnt = 0;	
	#ifdef ANALOG_ENCODER_ADC   //mechanical angle from encoder...ADC 4096 = 360 deg
	AngleLeft  =(adc_buffer.l_tx2*675)>>9 ;  // sf= (512*(360*15)/4096)/512= 675/512   conver to electrical phase equivalent
	AngleRight =(adc_buffer.l_rx2*675)>>9 ;  // right shifting >>9 divides by 512
	#else
	AngleLeft  =motAngleLeft ; // motor electrical phase 
	AngleRight =motAngleRight ;  
	#endif
	
	
	if(cnt == 0) //cnt is only used for debug and initialization
	{
		AngleLeftLast = AngleLeft ;
		AngleRightLast = AngleRight ;
	}
	diffL = AngleLeft - AngleLeftLast ;
	//if(cnt % 25 == 0 ) {printf( " diffL :%i diffR :%i  AngleLeft :%i AngleRight \r\n ",diffL , diffR,AngleLeft, AngeRight );}
	
if( diffL < -180 ) { cycleDegsL = cycleDegsL + 360; }
else 
if(diffL > 180) { cycleDegsL = cycleDegsL  - 360;}

	diffR = AngleRight - AngleRightLast ;
if( diffR < -180 ) { cycleDegsR = cycleDegsR + 360; }
else 
if(diffR > 180) { cycleDegsR = cycleDegsR  - 360;}

AngleLeftLast = AngleLeft;
AngleRightLast = AngleRight;
cnt++;
//************* convert to MotorPos cumulative electrical degrees
MotorPosL = (AngleLeft + cycleDegsL); // resolution 1 elec deg or 1/15  = .06666666667 mechanical deg ...15 poles on motor   
MotorPosR = (AngleRight +cycleDegsR);
  //if(cnt % == 0 ) printf( "MotPosR %i MotPosL %i \r\n" , MotorPosL,MotorPosR) ;
}

ducktrA, 2021-06-22:
Thank vamfun. So the best source is still the encoder. i was thinking the FOC is using an internal predictor which is backed by hall encoder. The logic is clear from there. Tx

vamfun, 2021-06-22:
Yes..only 90 counts/rev with hall sensors.   I’m using the analog output from Lamprey absolute  encoders which fit nicely between the the inside surface of the wheel and the axle mount.    We use these encoders in First robotics swerve drives.    The encoder does have a Uart output but I’m using the analog output currently.     I’ll put some test results with my PID Servo FOC fork soon.

http://vamfun.wordpress.com
Robot musing blog
iPhone

closed 2021-07-24 by EmanuelFeru