EFeru/hoverboard-firmware-hack-FOC #186: Hovercar self-brake at high speed

NikasAl, 2021-06-24:
### Your System
- VARIANT: HOVERCAR
- CONTROL TYPE: FOC_CTRL
- CONTROL MODE: TRQ_MODE
- What is your application?
    - Trike: Bicycle frame without brake + two drive wheels at the back

When the speed exceeds 30-35 km / h and I release the throttle stick, then for a short time there is a sharp braking. The faster I release the throttle and the higher the speed, the less often this self-braking.
This disturbs the stability of the trike and may fall, which is very dangerous.
At speeds less than 30 km / h, such self-braking does not occur and everything is fine.

This some of my Inc/config.h settings which isn't default: 
```c
//..
#define FIELD_WEAK_ENA  1
//..
//..VARIANT_HOVERCAR SETTINGS
//#define SIDEBOARD_SERIAL_USART3 1 
//#define FEEDBACK_SERIAL_USART3

#define PRI_INPUT1              1,  950, 2000, 3000, 0  // Pedal Brake        TYPE, MIN, MID, MAX, DEADBAND. 
#define PRI_INPUT2              1,  950, 2000, 3000, 0  // Pedal Accel        TYPE, MIN, MID, MAX, DEADBAND
//#define AUX_INPUT1              2, -1000, 0, 1000, 0  // Sideboard Steer    TYPE, MIN, MID, MAX, DEADBAND
//#define AUX_INPUT2              2, -1000, 0, 1000, 0  // Sideboard Speed    TYPE, MIN, MID, MAX, DEADBAND
//..
#define INVERT_L_DIRECTION                // Invert rotation of left motor
#define DEBUG_SERIAL_USART3 

```

EmanuelFeru, 2021-06-24:
This is a known issue when Field weakening is active... Need to look for a fix ...

mind999hacker, 2021-06-24:
I was having a very similar issue but opposite if I go on the throttle too fast it cuts out and breaks very intensively and the motherboard dies and I have to disconnect and reconnect the battery for it to work again. I have the field weakening off and I didn't mess around with any other settings and using a very similar setup as this any clue what could be going on? It seems like current protection although I might be wrong

NikasAl, 2021-06-25:
> I was having a very similar issue but opposite if I go on the throttle too fast it cuts out and breaks very intensively and the motherboard dies and I have to disconnect and reconnect the battery for it to work again.

This is what I mean too. But the motherboard does not turn off after installing the 4p10s battery instead 2p10s.

found the parameter 
`#define N_MOT_MAX 1000 // [rpm] Maximum motor speed limit`
 and set it a little more - at 1500 rpm, but have not yet experienced how this will affect self-braking. Probably that will not affect, because 1000 rpm is about 49 km / h on my wheels

EmanuelFeru, 2021-07-24:
Related to #164

closed 2021-07-24 by EmanuelFeru

farev, 2023-09-17:
> I was having a very similar issue but opposite if I go on the throttle too fast it cuts out and breaks very intensively and the motherboard dies and I have to disconnect and reconnect the battery [...]

I am having the exact same problem. Did you find a solution?