EFeru/hoverboard-firmware-hack-FOC #188: Problem with feedback USART3

Fiori23, 2021-06-29:
### Your System
- VARIANT: USART
- CONTROL TYPE: 
- CONTROL MODE:
- What is your application?
    - RC and autonomous Robot 

Goodevening Mr.Feru,

(My application has nothing to do with my problem because I'm using your original softwares).

I uploaded your firmware on the mainboard (variant USART). I only changed from USART2 to USART3 in the config.h file :

```h
// #define SIDEBOARD_SERIAL_USART2 0
 // #define CONTROL_SERIAL_USART2  0    // left sensor board cable, disable if ADC or PPM is used! For Arduino control check the hoverSerial.ino
 // #define FEEDBACK_SERIAL_USART2      // left sensor board cable, disable if ADC or PPM is used!

  // #define SIDEBOARD_SERIAL_USART3 0
   #define CONTROL_SERIAL_USART3  0    // right sensor board cable. Number indicates priority for dual-input. Disable if I2C (nunchuk or lcd) is used! For Arduino control check the hoverSerial.ino
   #define FEEDBACK_SERIAL_USART3      // right sensor board cable, disable if I2C (nunchuk or lcd) is used!
```
Then i uploaded your Arduino sketch without changing anything. It works: the motors spin but iI don't receive any serial feedback from the USART3.
Then i tried to use the debug mode in the config.h file to verify that it can read those feedback data:
```h
#define DEBUG_SERIAL_USART2 
```
And then connecting a USB-to-Serial FTDI adapter i can read the debug feedback.

Thanks in advance, in the hope of having been clear enought.

EmanuelFeru, 2021-06-29:
The feedback is in binary code, not ASCII. If you read it directly you will just see strange symbols. But if run the Arduino example, the Rx from the main board should be decoded for you if you look in the Arduino serial monitor.

Fiori23, 2021-06-29:
I tried this to read directly

```c
if (HoverSerial.available()) 
 Serial.println(HoverSerial.read()); 
```
but nothing
then I simply tried this and get only -1

```c
Serial.println(HoverSerial.read()); 
```

It seems to me that the mainboard doesn't send data

Candas1, 2021-06-30:
But is it working without the change from USART2 to USART3 ?

Fiori23, 2021-06-30:
I didn't try it but it should work with this simple change. The strange thing is that the Send() function in the Arduino sketch works perfectly but the Receive() funtion doesn't work. The mainboard doesn't send any feedback, only with the debug mode usart2 I can read the data.

Candas1, 2021-06-30:
I am asking because maybe you have a problem with USART3 Tx pin for example.

Fiori23, 2021-06-30:
Following your advice I just tried with USART2 and the Send() function works as well as USART3 but the recive function still not working. I also checked that my mainboard mounts a GD32F103RCT6 whitch has USART2 and USART3 5V tolerant.

Candas1, 2021-07-01:
Then it could be you have a problem from the arduino side, as you said you were reading debug feedback.
You could also try with the web tool mentioned [here](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Variant-USART) .

In Binary mode it should decode the feedback, it will prove the board and firmware work well at least.

Fiori23, 2021-07-01:
In tried to read in debug mode but only with a USB to Serial connected to USART2 and it works. The thing that doesn't work for me is only the Receive() function in USART2 and USART3.

vamfun, 2021-07-01:
Hi guys,
I've been following this thread since I too have had headaches getting the feedback uart3 tx signal to be read by my ESP32 which is basically derived from the Arduino sketch.  
 
I first thought my hoverboard USART3 tx wasn't working so wrote a stm32cube check program to verify booth USART2 and USART3 were functional.  

Still no FEEDBACK reads by Esp32.  So I bought this $13 logic Analyzer from Amazon. https://www.amazon.com/HiLetgo-Analyzer-Ferrite-Channel-Arduino/dp/B077LSG5P2 and it works great.   I verified that the signals from uart2 were working with FEEDBACK_SERIAL_USART2 set.  
Got no signal with FEEDBACK_SERIAL_USART3.  After several days of troubleshooting,I found a printf statement I used for debug without a #if define(debug_serial) ..#endif wrap around.  It never troubled me before since I always had a DEBUG_SERIAL_USART2 or ￼USART3 defined.  But in recent FEEDBACK testing I did not use debug so the printf  error showed up.

So finally verified that usart3 tX FEEDBACK was working with logic analyser.

But still FEEDBACK was not being read by￼ ESP32.   Finally wrote a tx/rx wrap test around for the Esp32 and found that the rx port had been fried by the hoverboard some how.   So rewrote the ESP32 program to use different rx/tx pins and finally was able to read/display the FEEDBACK from Usart3 tx!!

I'm using my FOC PID servo fork and soon will be able to fully test the usart3 control and feedback.  First with hall angle sensor and then with encoder sensor.   The encoder is mounted and ready to test.

Fiori23, 2021-07-01:
Thank you a thousand times. Following your advice i just checked with a logic analizer that the USART3 feedback works then I changed RX and TX pin from 2 and 3 to 10 and 11 and now it works.
Many thanks to everyone that helped me.

closed 2021-07-01 by EmanuelFeru