EFeru/hoverboard-firmware-hack-FOC #190: KZL-BC01-MAIN VER 05 Motherboard with GD32F103 Problem with power button

Kraxeeer, 2021-07-08:
Hello. I've already flashed foc firmware.
Almost stock, just foc and trq mode, also changed 
BOARD_VARIANT           0         // change if board with alternate pin mapping
From 0 to 1
The 1 variant is not working at all
The stock 0 variant is working when i power cpu from programmer (i hear beeps, it's blink led etc) but no power button reaction.
What now can i do with dead mb?
The KZL-BC01-MAIN VER 05 has 5pin halls and two addidionaly red and yellow connectors to, i think, provide 12V and 36V but was not connected in my board. CPU GD32F103
.
I have one dead motherboard with STM32F103 and about replacing stm to gd. i have soldering skills to do that.
[image]
[image]
Sorry for upside down model photo

EmanuelFeru, 2021-07-08:
If you only Power MCU and you hear beeps than MCU is ok. There might be something with the 2 TIP127 , check if they are not shorted, it always open. And the power latch pin from MCU to TIP127. Because somehow the MCU doesn't get power. So you have to narrow the problem. It can be the TIP127, then the 5V regulator, then the 3.3V regulator.

Kraxeeer, 2021-07-08:
The motherboard is fine. The problem appeared after flashing. This is my second flashed mb with your fw.

Candas1, 2022-01-02:
Can this be closed ? It's not a bug.
[Wiki](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Troubleshooting) has a few more details about what could be wrong on the board.

closed 2022-03-06 by EFeru

mhyargjsg-source, 2026-10-13:
Hi! I have a hoverboard with a KZL-BC01-MAIN Ver:05 motherboard. I would like to understand how the charging-detection circuit works and why the mainboard cannot operate while the charger is connected. Which components or signals control this function, and is there a manufacturer-approved way to operate the mainboard safely during charging, without disabling any safety protections? If possible, could you point me to the relevant section of the schematic? Thank you!