EFeru/hoverboard-firmware-hack-FOC #191: Is it possible to compile the firmware for GD32F303RCT6

peerke444, 2021-07-11:
Hi,

I've been playing with this firmware for a while with a lot of fun.
So I decided to buy another hoverboard to start a new project.
Unfortunately this motherboard has different processor: GD32F303RCT6.
The brand of the motherboard is TaoTao.
Is it possible to compile the firmware for this processor?

[image]
[image]

EmanuelFeru, 2021-07-11:
Hi, in not sure if it will work. To bed to do some research. First, to check the MCU datasheet, look at pin, memory and peripherals mapping. Then you have to check the pin mapping to the board, if they match with the firmware. 
I'm sure it's possible to make it work but maybe requires some effort.

peerke444, 2021-07-12:
Thanks for the tips.
I compared the datasheets and pin mappings.
Except from the fact one has a 108 mhz M3 processor and the other has a 120 mhz M4 processor I can't really find many differences.
Could you give me a hint how to convert your firmware to this processor?
I know my way in arduino but besides that...

Candas1, 2021-07-15:
Assuming this chip is fully compatible, you also need to check if the pins are mapped in the same [way](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility#pin-mapping)

peerke444, 2021-07-19:
That is a good question. 
Not really sure how to figure this out...

peerke444, 2021-07-20:
Ok, I started reverse engineer the PCB.
Most of the mapping I already found.
Some unfortunately not yet.
Below the list of mappings a have so far.
Let's assume I can also find the other mappings.
Could you tell me how to compile your software with the new mappings and for another processor?

This is the list I have so far: 
SWCLK = PA14
SWDIO = PA13
BUZZER = PC15
HALL LEFT U = PB8
HALL LEFT V = PB7
HALL LEFT W = PB6
HALL RIGHT U = PA8
HALL RIGHT V = PA12
HALL RIGHT W = PC10
LEFT UH = PC8
LEFT UL = PB1
LEFT VH = PC7 
LEFT VL = PB0
LEFT WH = PC6
LEFT WL = PA7
RIGHT UH = PA10
RIGHT UL = PB15
RIGHT VH = PA9
RIGHT VL = PB14
RIGHT WH = PA8
RIGHT WL = PB13

EmanuelFeru, 2021-07-24:
Please have a look at the Driver files for your MCU. Generate a project using MXCube for your MCU, because youmight need completely new HAL_ files.
Would be ideal that the same HAL_files are used for your MCU, the you only need to change the pin definitons in [defines.h](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/Inc/defines.h)

peerke444, 2021-08-29:
Played some days with CubeMX.
But I think this is a bit to complex for me unfortunately...
Thanks for your help so far!

closed 2022-03-06 by EFeru

LeonXBB, 2024-08-20:
I have made the following table
[image]
I noticed A8 has two functions. Is it possible or is it a mistake?