EFeru/hoverboard-firmware-hack-FOC #192: Hoverboard model

rdignoti, 2021-07-14:
Can you provide me a link for the hoverboard model that have the same motherboard that you are using?

Thanks

Candas1, 2021-07-14:
Hi. 

It's not clear what hoverboard has what motherboard unfortunately.
I think we had better luck with older hoverboards. 
To drive the cost down, they came up with hoverboards with different mainboards, different chips, cheaper wheels, smaller batteries.

[Wiki](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility) explains what boards are compatible with this firmware.

rdignoti, 2021-07-14:
Thank you, seems like a russian roulette buy a brand new hoverboard, much better second-hand ones

Candas1, 2021-07-14:
I bought many second-hand ones, and got a lot of split boards.
So you can sell it if you are not happy with the board inside.

closed 2021-07-24 by EmanuelFeru