EFeru/hoverboard-firmware-hack-FOC #193: Anti-theft wheel motor lock mode.

2007-i, 2021-07-16:
1. Please tell me how to make the wheel motor lock mode for the proto-theft system?
2. Is it possible to turn the Hovercar on and off by giving a logical 1 and 0 (1 - turn on, 0 - turn off)? It would also be useful when using the anti-theft function.
Now many cheap anti-theft systems are sold and many make mini buggies, which, when the wheel is completely locked, will be very difficult to move if they want to steal it))
I think the controller has unused inputs for this and maybe you just need to add a code to block and turn off the system when a signal is given?
Thanks in advance for your reply and for your great work!

2007-i, 2021-07-17:
Does this line serve this purpose?
If you uncomment it, will the wheel be locked only at rest?

// #define STANDSTILL_HOLD_ENABLE            // [-] Flag to hold the position when standtill is reached. Only available and makes sense for VOLTAGE or TORQUE mode.

EmanuelFeru, 2021-07-18:
Yes, that will hold position, so it can be used as locking feature for short period, like going into a shop. The drawback however with such a method is, that the whole electronics need to be powered on.
Maybe a more useful locking method is to use a switch with a key that shorts the motor phases together. And the same switch turns off the power to the board. So the functionality will be something like this:
- switch off: power off, motors phases shorted
- switch on: power on, motors phases not shorted ( connected to main board)

2007-i, 2021-07-18:
Yes, also an option!

#if defined(STANDSTILL_HOLD_ENABLE) && (CTRL_TYP_SEL == FOC_CTRL) && (CTRL_MOD_REQ != SPD_MODE)
if (!rtP_Left.b_cruiseCtrlEna) { // If Stanstill in NOT Active -> try Activation
if (((input1[inIdx].cmd > 50 || input2[inIdx].cmd < -50) && speedAvgAbs < 30) // Check if Brake is pressed AND measured speed is small
|| (input2[inIdx].cmd < 20 && speedAvgAbs < 5)) { // OR Throttle is small AND measured speed is very small
rtP_Left.n_cruiseMotTgt = 0;
rtP_Right.n_cruiseMotTgt = 0;
rtP_Left.b_cruiseCtrlEna = 1;
rtP_Right.b_cruiseCtrlEna = 1;
standstillAcv = 1;
}

It seems that it is possible and this condition can be applied so that it would work only in rest mode.
I just didn’t figure out how to launch it correctly.

2007-i, 2021-07-18:
Is there any controller input that can be used for the condition? And we will have, just like electric scooters, a special entrance for the anti-theft device.
I'm ready to solder the contact on the controller if you need it.