EFeru/hoverboard-firmware-hack-FOC #194: unable to get the correct output from /dev/ttyTHS1

darld, 2021-07-20:
connect mainboard to jetson nano via GPIO port 8 & 10 (RX & TX)
- VARIANT: USART
- What is your application?
    - send command to mainboard to control the motor
   
> desktop@desktop:~$ cat /dev/ttyTHS1 | hexdump
0000000 00ff ff00 0000 00ff 0000 00ff 0000 00ff
0000010 0000 00ff 0000 ff00 0000 00ff ff00 0000
0000020 00ff 0000 00ff ff00 0000 ff00 0000 ff00
0000030 0000 ff00 0000 ff00 0000 ff00 0000 ff00
0000040 0000 ff00 0000 ff00 0000 ff00 0000 00ff
0000050 0000 00ff 0000 00ff 0000 00ff 0000 00ff
*
0000070 0000 00ff 0000 00ff 0000 00ff ff00 0000
0000080 00ff 0000 00ff ff00 0000 00ff ff00 0000
0000090 ff00 0000 00ff ff00 0000 00ff 0000 00ff
00000a0 ff00 0000 00ff ff00 0000 00ff ff00 0000
00000b0 00ff 0000 00ff ff00 0000 00ff ff00 0000
00000c0 0000 00ff 0000 00ff 0000 00ff ff00 0000
00000d0 00ff ff00 0000 00ff ff00 0000 00ff ff00
00000e0 0000 00ff ff00 0000 ff00 0000 00ff ff00
*
0000100 0000 00ff ff00 0000 00ff ff00 0000 00ff
0000110 ff00 0000 ff00 0000 ff00 0000 ff00 0000
*
0000130 ff00 0000 00ff 0000 00ff 0000 00ff 0000
0000140 00ff 0000 00ff 0000 00ff 0000 00ff 0000
0000150 00ff ff00 0000 00ff 0000 ff00 0000 00ff
0000160 ff00 0000 00ff ff00 0000 00ff ff00 0000
0000170 00ff ff00 0000 00ff ff00 0000 00ff ff00
0000180 0000 00ff ff00 0000 00ff ff00 0000 00ff
0000190 ff00 0000 00ff ff00 0000 00ff ff00 0000
00001a0 00ff ff00 0000 00ff ff00 0000 00ff ff00
00001b0 0000 00ff ff00 0000 00ff ff00 0000 00ff
00001c0 ff00 0000 00ff ff00 0000 00ff ff00 0000
00001d0 00ff ff00 0000 00ff ff00 0000 00ff ff00
00001e0 0000 00ff ff00 0000 00ff ff00 0000 ff00
00001f0 0000 00ff ff00 0000 00ff ff00 0000 00ff
0000200 ff00 0000 0000 00ff 0000 00ff 0000 00ff
0000210 0000 00ff 0000 00ff 0000 00ff 0000 00ff
0000220 ff00 0000 00ff 0000 00ff 0000 00ff 0000
0000230 00ff 0000 00ff ff00 0000 00ff ff00 0000
0000240 00ff 0000 00ff 0000 ff00 0000 00ff ff00
0000250 0000 ff00 0000 00ff ff00 0000 00ff ff00
0000260 0000 ff00 0000 00ff 0000 00ff 0000 00ff
0000270 0000 00ff 0000 00ff 0000 00ff 0000 00ff
0000280 0000 00ff 0000 00ff 0000 00ff 0000 ff00
0000290 0000 00ff ff00 0000 ff00 0000 00ff 0000

Expect the output will contains the form like 0xabcd as describe in the main page. But I get this weird format. Also, unable to using the candas1's web tools to visualize the output, but I can see the I/O steaming. 
The method I used for flashing is platformIO and the flashing is successful. 
Thus, I not sure what reason that I will get this weird format as above.

EmanuelFeru, 2021-07-21:
Can you give more details on how do you send the UART values from Jetson Nano to main board? 
Please also check the Arduino example or the Sideboard repo to see how it is done.

PuceBaboon, 2021-07-27:
Hi Darld,

   First, let me make it plain that I do **not** have this problem.

   However, I'm using the USART version (in my case, connecting to an ESP32) and this pattern is exactly what I see when I'm testing the ESP32 on the bench, disconnected from the Hoverboard motherboard (so the flying lead for the USART connection is sticking up from the ESP32 board, but not actually connected to anything).

   So what you're seeing is just random line noise and you almost certainly have a hardware connection problem.  Double check that you're using the correct port (on both ends) and make sure that there's a good ground reference between the two boards (**don't** plug the charger into the Hoverboard when you also have the Jetson running from a mains PSU).

   Good luck!

                      -John-

closed 2021-07-28 by darld