EFeru/hoverboard-firmware-hack-FOC #196: flash with linux cli

darld, 2021-07-23:
when flash with linux cli and run make, it return this error

`desktop:~/hoverboard-firmware-hack-FOC$ make
mkdir -p build
arm-none-eabi-gcc -c -mcpu=cortex-m3 -mthumb   -DUSE_HAL_DRIVER -DSTM32F103xE -IInc -IDrivers/STM32F1xx_HAL_Driver/Inc -IDrivers/STM32F1xx_HAL_Driver/Inc/Legacy -IDrivers/CMSIS/Device/ST/STM32F1xx/Include -IDrivers/CMSIS/Include -Og -Wall -fdata-sections -ffunction-sections -std=gnu11 -g -gdwarf-2 -MMD -MP -MF"build/stm32f1xx_hal_flash.d" -MT"build/stm32f1xx_hal_flash.d" -Wa,-a,-ad,-alms=build/stm32f1xx_hal_flash.lst Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_flash.c -o build/stm32f1xx_hal_flash.o
/bin/sh: 1: arm-none-eabi-gcc: not found
Makefile:152: recipe for target 'build/stm32f1xx_hal_flash.o' failed
make: *** [build/stm32f1xx_hal_flash.o] Error 127
`

EmanuelFeru, 2021-07-24:
I see you are missing the "arm-none-eabi-gcc: not found", so you might need to do:
```
apt install gcc-arm-none-eabi
```
as mentioned here: https://unix.stackexchange.com/questions/377345/installing-arm-none-eabi-gcc

opendeliveryrobot, 2021-07-27:
desktop:~/hoverboard-firmware-hack-FOC$  make flash
st-flash --reset write build/hover.bin 0x8000000
st-flash 1.7.0
st-flash: symbol lookup error: /usr/lib/libstlink.so.1: undefined symbol: libusb_set_option
Makefile:179: recipe for target 'flash' failed
make: *** [flash] Error 127

Get this error even I have install the st-link utility.

closed 2022-03-06 by EFeru