EFeru/hoverboard-firmware-hack-FOC #197: problem with USART GD32F103

olegi4a, 2021-07-24:
I use standard firmware with the following settings:
#define VARIANT_USART // Variant for Serial control via USART3 input
#define CONTROL_SERIAL_USART3
#define FEEDBACK_SERIAL_USART3
// #define DEBUG_SERIAL_USART3 - commented out

the board does not squeak!!! using the site for UART control, all data coming from the board are displayed (speed, battery voltage ...).

PROBLEM - the board does not respond to the joystick from the site, just does not respond to the Arduino with the control firmware.
  on usart2 just the same problem.

I have two more boards on STM32F103 - everything works perfectly !!!

EmanuelFeru, 2021-07-24:
Can you post a picture with your GD board?

olegi4a, 2021-07-24:
the board works flawlessly in ADC mode

> Можете ли вы опубликовать картинку с помощью доски GD?

[image]
[image]

Candas1, 2022-01-02:
Hi,

Can this be closed ?
ADC is on the left wire, you are using USART3 so the right wire. It could be your RX pin on USART3 is defective.
You could try USART2 instead.
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-USART

closed 2022-01-17 by Candas1