EFeru/hoverboard-firmware-hack-FOC #198: [feature request] current slushing

goosst, 2021-07-24:
Hello 

When I enable the hoverboard and request zero torque (through uart), I still hear the current slushing.
Since I like to let the inverter turned on for a long time, it has some negative effect on battery usage (although I can't immediately prove it of course).

I've tried going through the code but didn't find it obvious to find the flow from the steering/speed commands to dutyratio of the switches. 
Normally I would just edit the simulink model and set pwm to zero at zero torque request (and potentially only at zero speed). But I don't have the tools to generate the code out of it, manually adapting is also not the best idea if it would ever get merged in the main branch.

I believe to understand `pwml;` (main.c), gets written by the received (uart-)commands, is an input to the matlab function and has actually no relation with the applied pwm to the switches. So if sending this a zero in torque mode, it would regulate to zero current. Some control action of the current loop to control to zero could explain some audible noise and continuous switching.

I also see an `enable` being set in main.c (at overtemperatures etc.). Would I be able to add some logic around zero torque to toggle this enable and still get power to the halls? I'm still using the 15V of the halls to power an ESP32, which I want to keep turned on. 
Or do you have other advice where to add such a piece of logic?

thanks!

EmanuelFeru, 2021-07-24:
Yes, at 0 torque request, current is being controlled, that is why you can hear it if it is very silent around you.
And indeed, you could put the enable to 0 after a certain time at 0 speed. Hall sensors and corresponding speed will still be calculated.
You if course have to think how do you put back enable to 1, depending on your situation.

goosst, 2021-07-30:
Thanks, implemented it. 
I find it actually amazing how much the standy power consumption is of the board without doing any actuation (or my esp32 must be sucking a lot of power).

Candas1, 2022-01-02:
@goosst I believe you have a working solution, closing this issue

closed 2022-01-02 by Candas1