EFeru/hoverboard-firmware-hack-FOC #200: Current sensing possible?

PaddyCube, 2021-07-29:
### Your System
- VARIANT: USART2
- CONTROL TYPE:
- CONTROL MODE:
- What is your application?
    - robot lawn mower

Hello and thanks for this cool Firmware. 
Is it possible to get current sense of each Motor in mA or A? Meaning how much current gets drawn. I'm not familiar with bldc motors but I used this to detect overload/blocking situation and to shut off.
So is there a way to limit the Power of motors to 100W or so by sense the current?

Patrick

NickLD, 2021-08-02:
@PaddyCube
You can specify a current limit (per motor) in the config.h:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/eb20cc0acad0da91ead7d207fa6a83f1778f03e1/Inc/config.h#L154
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/eb20cc0acad0da91ead7d207fa6a83f1778f03e1/Inc/config.h#L155

The Config comments should guide you well on what to set for them :)

PaddyCube, 2021-08-03:
Thank you for this hint. What happen if one of these limits gets reached? Does it stop, stop and go to error state, turn off or what? Also can we recover from such a situation? I'm using USART commands to control the robot.

NickLD, 2021-08-03:
As the `I_DC_MAX` define states, "Above this value, current chopping is applied."
So, in the event motor attempts to pull more than that amount of current, it will apply 'current chopping' which effectively limits the current.

CarlDavey, 2022-08-09:
I would find this useful also. I have a 4wd drive configuration using a teensy 4.0 and the Arduino example. For control I have a RC radio with two way telemetry via FrSky S.Port. A readout of current consumption on the radio display would be amazing.

In the Arduino Sketch feedback is currently:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/eb20cc0acad0da91ead7d207fa6a83f1778f03e1/Arduino/hoverserial/hoverserial.ino#L129

In the comms.c source code there is reference to the current consumption variables:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/eb20cc0acad0da91ead7d207fa6a83f1778f03e1/Src/comms.c#L132

Can I simply add some additional variables to the feedback functions?

And an awesome project, thank you.

Candas1, 2022-08-10:
You can check what I did here
https://github.com/Candas1/hoverboard-firmware-hack-FOC/commit/2036a3e4e7feaf64ce18bd4edfed2c6108f66035

CarlDavey, 2022-08-10:
> You can check what I did here [Candas1@2036a3e](https://github.com/Candas1/hoverboard-firmware-hack-FOC/commit/2036a3e4e7feaf64ce18bd4edfed2c6108f66035)

Nice one, that will work. I will also change the struct and checksum calculation in my arduino sketch.

closed 2022-12-23 by Candas1