EFeru/hoverboard-firmware-hack-FOC #201: Motor shaking at low speed

LoneRetrievr, 2021-08-02:
Hello,
I'm trying to control the motors with an Arduino. It works well despite the fact that both motors are shaking at low speeds. I tried both ADC and PWM variants with my custom program :
```
void setup() {
  pinMode(10, OUTPUT);
  pinMode(11, OUTPUT);

}

void loop() {

  for (int i=0; i<=255; i = i + 40) {
    analogWrite(10, i);
    analogWrite(11, i);
    delay(2000);
  }

}
```

The goal with this code was just to understand how everything worked, and to see how I could control speed. As the speed is increasing, the vibrations are reducing, but they're very violent at the beginning. I also tried the program included with the firmware in USART mode and I had the same problem. I don't think it comes from the motors, because the hoverboard was working perfectly. Do you have any clue ?
Thanks a lot

Candas1, 2021-08-02:
Hi, 
Have you tried this
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility#troubleshooting

LoneRetrievr, 2021-08-02:
I just tried this and nothing good happened. The motors either didn't work at all or span way too fast with a burning smell... Thankfully they still work

kris86k, 2021-08-28:
Try to swap motor cable for example green to yellow.

LoneRetrievr, 2021-09-01:
I have already tried this. It didn't work and the reactions I got from the motors were scary, so I won't try this again.

villamany, 2021-12-17:
Hi, I got some issues, some times one motor doesnt start or start if i move mnually first, also the "beep, beep" on back doesnt work.
I modified these lines in "define.h" and now works fine, maybe a different hall wiring on mine, but works.

#define LEFT_HALL_U_PIN GPIO_PIN_5//GPIO_PIN_5
#define LEFT_HALL_V_PIN GPIO_PIN_7//GPIO_PIN_6
#define LEFT_HALL_W_PIN GPIO_PIN_6//GPIO_PIN_7

#define LEFT_HALL_U_PORT GPIOB
#define LEFT_HALL_V_PORT GPIOB
#define LEFT_HALL_W_PORT GPIOB

#define RIGHT_HALL_U_PIN GPIO_PIN_10//GPIO_PIN_10
#define RIGHT_HALL_V_PIN GPIO_PIN_12//GPIO_PIN_11
#define RIGHT_HALL_W_PIN GPIO_PIN_11//GPIO_PIN_12

AdamMulica, 2022-03-20:
Hello,

I have a similar problem with my hoverboard chip GD32F103RCT6. I am using a potentiometer for the speed of the motors. The right engine works without any problems. The left engine jumps. Sometimes it stutters and when I move it manually it works, but it jumps and you can feel it warming up on the engine housing. Both motors turn at the same speed. Where to look for the cause? I used different variants but the effect is the same. If I swap the wires, the effect is the opposite.

film
https://drive.google.com/file/d/1afbR3y8EY9FpqpSUjyKm4EqFhkLo_Ciw/view?usp=sharing

Regads.

Candas1, 2022-03-21:
Have you tried to swap the motors ?

AdamMulica, 2022-03-21:
When I changed the wires from the controller alternately to the motors, the effect was the opposite. So the right one was vibrating and the left one was spinning well. And after a few hours of fighting with the software etc. I damaged the chip GD32F103RCT6 UPSSSS

Candas1, 2022-03-21:
So the problem is with the motors right ?

AdamMulica, 2022-03-21:
Not ! motors are ok! Motherboard problem...

closed 2022-05-11 by Candas1