EFeru/hoverboard-firmware-hack-FOC #202: Making speed limit for hoovercar

jusubbi, 2021-08-04:
Hello,

I'm using VARIANT_HOVERCAR for my DIY electric scooter and it's working fine. Now i'm modified tires 8" -> 12,5" and it goes way too fast. There is max rpm setting for motors, but is it possible to change double tap reverse -> speed limiter?

I meant normally speed limit is like 500rpm and if i want go faster i double tap brake and it will change limit to 1000rpm.

Is this possible? I'm not a coder :(

Candas1, 2021-08-04:
Have you checked this ?
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Setup-current-and-speed-limits

jusubbi, 2021-08-04:
Yes i have. Those are for permanent current / speed limit.

Candas1, 2021-08-04:
No it's not. As you can change it again with same procedure.

closed 2021-12-27 by Candas1