EFeru/hoverboard-firmware-hack-FOC #203: how to count wheel rotations and find position from usart serial

2by4mike, 2021-08-04:
### Your System
- VARIANT: USART
- CONTROL TYPE: FOC_CTRL
- CONTROL MODE: VLT_MODE 
- What is your application?
  moving cooler robot

**Indicate what you have tried:**
I have tried looking through the code to identify where I could insert a counter to count the hall sensors (peak) readings and where I put them to be read by the esp, but I don't understand where it would be put, and now that I think about it how I would do it.

so what should I do to add wheel position trackers?

EmanuelFeru, 2021-08-04:
The motor controller outputs the electric angle for each motor, search in `bldc.c` for `motAngleLeft = rtY_Left.a_elecAngle;` same for Right motor.
Maybe it help if you need motor position.

Candas1, 2021-08-04:
This fork seems to evaluate and send wheelcount for robotic usage
https://github.com/alex-makarov/hoverboard-firmware-hack-FOC

mczubel, 2021-11-11:
In this fork https://github.com/alex-makarov/hoverboard-firmware-hack-FOC  5 and 6 values in UART feedback are R and L wheel count. Calculate  rpm, distance, etc. I work with python in raspberry but is similar for esp32 micropython.

closed 2022-01-17 by Candas1