EFeru/hoverboard-firmware-hack-FOC #206: Nunchuck not working

WilliamAnderssonDev, 2021-08-11:
### Your System
- VARIANT: Nunchuck
- CONTROL TYPE: 
- CONTROL MODE:
- What is your application?
    - Golf Trolley

Hello, i have connected a Nunchuck like described in https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Variant-NUNCHUK 

It beeps like normal when i start and i have no error beep code but when i move the nunchuck nothing happens and its official Wii nunchuck

Candas1, 2021-08-11:
Hi,

Have you maybe wrongly entered auto-calibration mode and didn't proceed properly ?

WilliamAnderssonDev, 2021-08-11:
How do i do that?

Candas1, 2021-08-11:
Did you read the wiki ?
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Variant-Nunchuk

WilliamAnderssonDev, 2021-08-11:
Yes i have, i tried to autocalibrate now and still nothing

Candas1, 2021-08-11:
No error beeps ?
How is your mainboard looking like ?

WilliamAnderssonDev, 2021-08-11:
Please check my video here https://www.youtube.com/watch?v=jyXJUdMHx0g

Candas1, 2021-08-11:
Can you also record on video how you do the auto-calibration please ?

WilliamAnderssonDev, 2021-08-11:
Ok, i did flash erase to remove auto calibration, should i calibrate again?

Candas1, 2021-08-11:
Auto-calibration is nice to have with nunchuk to have joystick properly centered, but it's not mandatory.
Nunchuk variant should work out of the box even without auto-calibration if you have proper wiring.
I tried it myself many times.

So it's still no working after erasing and flashing ?

WilliamAnderssonDev, 2021-08-11:
Nope the video is my latest step

WilliamAnderssonDev, 2021-08-11:
Nunchuck SDA(green) is connected to hoverboard green. Nunchuck red is to 3.3v, Nunchuck white is to hoverboard black and nunchuck SCL(yellow) is to board blue. And i have resistor like said. I connected everything to the right sideboard as described in this picture. https://raw.githubusercontent.com/EmanuelFeru/hoverboard-firmware-hack-FOC/master/docs/pictures/mainboard_pinout.png

Candas1, 2021-08-11:
Are you sure you picked the right variant as mentioned [here](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Variants) ?
How are you compiling and flashing ?

WilliamAnderssonDev, 2021-08-11:
Yep, im using platformio.
platformio.ini: https://pastebin.com/DtYrCHDE
config.h: https://pastebin.com/VqPWQwGL

WilliamAnderssonDev, 2021-08-11:
i use st link v2 to flash with STM32CubeProgrammer

WilliamAnderssonDev, 2021-08-11:
I have tried the arduino with USART too but i just get the 3 beeps so i dont understand what im doing wrong

Candas1, 2021-08-12:
Maybe you have problems with the MCU ports on the right cable which could also explain why nunchuk is not working, you can try USART2 and left cable
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Variant-USART

WilliamAnderssonDev, 2021-08-12:
Doesnt arduino destroy usart2 as it outputs 5v?

Candas1, 2021-08-12:
only usart3 ones are 5v tolerant
[image](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/docs/pictures/mainboard_pinout.png)

WilliamAnderssonDev, 2021-08-12:
Then i can't try usart2 if i cant get nunchuck there. I dont know why nothing works for me because the hoverboard was working when i took it apart.

Can you help me with config.h and platformio.ini for nunchuck use like you had
and maybe one for arduino so im sure that its not my config thats wrong

Candas1, 2021-08-13:
Well apart from selecting the right variant, there shouldn't be anything to change
Do you have a ftdi to check in ascii debug what input values look like?

WilliamAnderssonDev, 2021-08-13:
I dont have one. I just dont understand why nothing works because it was a fully functional hoverboard

Candas1, 2021-08-14:
Maybe you are doing something wrong, or maybe your board is different.

As you can see firmware works
https://youtu.be/SYkKr-bOYis

WilliamAnderssonDev, 2021-08-16:
My board is same as yours

vamfun, 2021-08-18:
You can try usart2  with a voltage divider.   Put 5 v across pot and adjust center to 3.3 v .   Lock pot with tape and then put Arduino across pot and connect to usart2 .

1a2s3d4f5TDS, 2022-01-28:
Hi everybody i have a same situation as topic starter
No beeps no spin. when i try disconnect signal wire blue or green i got error beeps. One more important thing. My board variant 1 (usual 0)

SerjeoWN, 2022-01-28:
> ### Your System
> * VARIANT: Nunchuck
> * CONTROL TYPE:
> * CONTROL MODE:
> * What is your application?
>
>   * Golf Trolley
>
> Hello, i have connected a Nunchuck like described in [...]

Hi everybody i have a same situation as topic starter
No beeps no spin. when i try disconnect signal wire blue or green i got error beeps. One more important thing. My board variant 1 (usual 0)

SerjeoWN, 2022-01-28:
Hi everybody i have a same situation as topic starter
No beeps no spin. when i try disconnect signal wire blue or green i got error beeps. One more important thing. My board variant 1 (usual 0)

Candas1, 2022-01-28:
You probably did a bad calibration by mistake, either do a proper calibration or erase/flash the chip again. Check the details in the wiki.

SerjeoWN, 2022-01-29:
> You probably did a bad calibration by mistake, either do a proper calibration or erase/flash the chip again. Check the details in the wiki.

[https://www.youtube.com/watch?v=5dqO9VF2zsc](url)

1) no second sound after turn on . only after when im unplug nunchuck 2) when i play with Nunchuck plug , i can run motor , but until next off/on

Candas1, 2022-01-30:
Have you calibrated as I mentioned?
Motors won't enable at startup if the inputs are not very small. So if your joystick is not well centered it won't work. That's why you need to calibrate the inouts either manually with debug serial or the auto calibration.

SerjeoWN, 2022-01-30:
> Have you calibrated as I mentioned?

> 

https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Input-Calibration
yes
Auto-calibration finds the minimum, middle and maximum values for the inputs. It also identifies the type of input and checks if it can be protected (ADC only).

Turn the board on and press the power button for more than 5 sec and release after the beep sound
Move the potis freely to the min and max limits repeatedly
Release potis to the resting position
Press the power button to confirm or wait for the 20 sec timeout

> Also as mentioned in the wiki you should better use pull up resistors.

yes. 91 kOm Blue-Groud and Green-Ground

i think i catch my issue
Please check my video 
https://www.youtube.com/watch?v=nKbINWaS1-o

Candas1, 2022-01-30:
I will try to reproduce it at home tomorrow

Candas1, 2022-01-31:
OK, I found a very weird bug, and it confirms what @WilliamAnderssonDev, sorry I didn't take it seriously back then.
I think it might not be happening on all boards, and the board I used now has the same issue. Because I had tested nunchuk in the past and there was no issue.

@SerjeoWN, can you please try following change, and if confirmed I will do a pull request.
[Here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/4770b13b2cfc3a952f17f6217ec254639d10a0f5/Src/util.c#L293) in your util.c file, please add the following command `HAL_Delay(100);`

Before:
```
I2C_Init();
Nunchuk_Init();
```

After:
```
I2C_Init();
HAL_Delay(100);
Nunchuk_Init();
```

SerjeoWN, 2022-01-31:
> I found a very weird bug, and it confirms what @WilliamAnderssonDev, sorry I didn't take it seriously back then.

Thank you. It seemed to me that I was going crazy and losing faith in my knowledge and the power of the mind.

> I think it might not be happening on all boards, and the board I used now has the same issue.

Does your board have an alternative pin configuration? 1 not 0 ?
maybe this is a problem alternative boards ?

> can you please try following change, and if confirmed I will do a pull request.

I'll test it for a few days and let you know the result.

Candas1, 2022-01-31:
No it's not an alternative board.
I am not sure I am confused now. This delay fixed the issue for me. Looks like we try to initialize the nunchuk too quickly after the initialization of I2C.
But there is a piece of code that is supposed to initialize the nunchuk again after few failures. Maybe we broke that piece of code during last change in december 2020... I would need to try older versions of the firmware to be sure.

SerjeoWN, 2022-02-01:
> like we try to initialize the nunchuk too quickly after the initialization of I2C.
I think so too.
1)The nunchuk gets power quickly.
2)But the hover board hasn't booted yet
3)Nunchuk does not receive a response from the board, and goes to sleep
4)The board has loaded and does not find the nunchuck (he is sleeping)
5)error.

Candas1, 2022-02-01:
The weird thing is it's always working after flashing.
So if I try debugging from platformio, I cannot reproduce it as it's first flashing the firmware.
I will do more tests in the evening.

In the meantime please let me know if the workaround fixed the issue for you.

SerjeoWN, 2022-02-01:
> The weird thing is it's always working after flashing.

not always. Often I have to upload another firmware. I suspect that the flash memory is not completely erased, even when I manually erase the memory through the STM software.

Candas1, 2022-02-01:
This is only when calibration is bad and saved to eeprom.
But I checked it, it happens even without the calibration in eeprom.

Candas1, 2022-02-01:
I tried a version from October 2020, still couldn't get Nunchuk to work. Almost seems like it was never working, although I have recorded videos where it's working.
Only this delay helps.

I am wondering now if it could be due to an update in the toolchain, resulting in a different binary.

SerjeoWN, 2022-02-03:
> In the meantime please let me know if the workaround fixed the issue for you.

yes and no.

here's some more video with my terrible english.  ))))
https://www.youtube.com/watch?v=BZu9w7EcN20&t=24s
and looks like i found another bug
https://www.youtube.com/watch?v=nKbINWaS1-o&t=5s

Candas1, 2022-02-03:
From your video, I cannot understand what is not working.
Can you rephrase it here please ?

SerjeoWN, 2022-02-03:
> From your video, I cannot understand what is not working. Can you rephrase it here please ?

after the first turn on, the Nunchuk  does not initialize, there is no second tune, until you start moving the joystick.

Candas1, 2022-02-03:
But is it without calibration ? 
Can you check the debug serial output with a ftdi ?

There is a security feature that will enable wheels (second tune) only if the values are really small (-50 to 50).
If you joystick is a bit off center it can happen.

sdegeorgio, 2022-02-06:
I've also had big problems getting the nunchuck to work.  I've studied the code and made quite a few changes to improve the connectivity.  It took lots of scoping on the I2C bus to ensure reliable bus re-starts in event of errors...  I hope it may help improve the experience for others too.
Stephan.

Candas1, 2022-02-06:
Hi Stephan,
I tried your changes now, it definitely does a much better job when unplugging/plugging the nunchuk. 
I also tried Dual inputs with USART and NUNCHUK, it worked well. 

Thank for the great contribution.
Adding @EFeru for visibility has he his the owner of the repository.

Candas1, 2022-02-22:
@SerjeoWN, have you tried it as well? Any feedback?

EFeru, 2022-03-06:
Should be fixed by #260

closed 2022-03-06 by EFeru