EFeru/hoverboard-firmware-hack-FOC #208: Control each motor individually

Henrado, 2021-08-12:
### Your System
- VARIANT: UART
- CONTROL TYPE: FOC_CTRL
- CONTROL MODE: VOLTAGE MODE
- What is your application?
    - Arduino controlled robot

I'm checking the hoverboard with an arduino, this works great using hoverserial.ino.
But:
Is it possible to disable the mixer? So that one can use something like: send (wheel 1 speed, wheel 2 speed) insted of send (stear, speed)?

This is so that I can more easily control each motor individually.

Thank you in advance! You have done a great job

Candas1, 2021-08-12:
Hi,
You can check this
https://github.com/labrats-x7/hoverboard-firmware-hack-FOC/commit/eea5af8deb34350907c21a9ecd53411c79b00102

Henrado, 2021-08-12:
Exactly what I was looking for! Thank you very much

closed 2021-08-12 by Henrado