EFeru/hoverboard-firmware-hack-FOC #209: Adc not working

WilliamAnderssonDev, 2021-08-17:
Hello, i can't get adc to work. Iam using 2 potentiometers and i know its not a hardware issue because it works when i use Niklas Fauth's firmware.

Can someone help me with a config file, i really dont need steering i just need to move forward with 1 10k pot

EmanuelFeru, 2021-08-17:
You have to connect both Potis otherwise you get a timeout. If you only want to use one poti then you have to disable one channel in the firmware. You need to have a look in the code.

WilliamAnderssonDev, 2021-08-17:
I have connected both but it doesnt work

WilliamAnderssonDev, 2021-08-17:
I dont think it detects them, they are connected like this 
[image]

vamfun, 2021-08-17:
Might be a calibration issue.      Do a memory erase and repeat the full calibration procedure.

http://vamfun.wordpress.com
Robot musing blog
iPhone

WilliamAnderssonDev, 2021-08-17:
Hello now the board doesnt turn on at all, not by power and nothing happens when i connect programmer

WilliamAnderssonDev, 2021-08-17:
It worked, then i power everything off and disconnect battery and went for food and now when i plug everything back in it doesnt work

vamfun, 2021-08-17:
verify that 3.3 v is present on programming plug…if that’s ok take a look at https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/issues/129

This saved me😎

http://vamfun.wordpress.com
Robot musing blog
iPhone

WilliamAnderssonDev, 2021-08-18:
No voltage

vamfun, 2021-08-18:
Make sure you have battery voltage on board input.   Hold the switch in on position and measure 3.3 pin.   If still no 3.3 then possible switch problem or power circuits. Replace switch with jumper and retry 3.3 v measurement.

http://vamfun.wordpress.com
Robot musing blog
iPhone

WilliamAnderssonDev, 2021-08-18:
I have tested all and still nothing so i will buy a new board

Candas1, 2022-01-02:
Hi, any updates ? Can this issue be closed ?

closed 2022-03-06 by EFeru