EFeru/hoverboard-firmware-hack-FOC #210: More precise torque control at low speeds with external encoder

Hylke33, 2021-08-22:
So my idea is to use a hoverboard motor to create a direct drive race wheel. In the basics I got it working. I use an ATMEGA32u4 to connect to the PC as an Joystick device and i'm getting the force feedback values from the game. I then send them to the motor controller via UART as an constant torque value. Initially I wanted to use the internal encoder of the hoverboard motor as input, but this turned out not to be precise enough. I therefore added an external encoder to precisely measure the wheels position for a much smoother gaming experience.

When the wheel is turning, the constant torque is great, but when the steering wheel turns very slowly and with very small corrections (when driving on the straights), you can feel the motor controller is less capable of keeping a constant torque and you can feel small "bumps" in the force. Could the constant torque resolution be improved by using the external encoder so the motor controller knows more precisely what coils to power when or is that not at all possible and just an limitation of using this motor?

fluppie, 2021-08-22:
Have you seen this repo? Maybe it can be helpful? What kind of encoder is attached?
https://github.com/vamfun/Hoverboard-Hack-FOC-PID-Servo

Hylke33, 2021-08-23:
Hey Fluppie, 

Thanks! I did see it flashing by, I did not see they also used an external encoder to drive the motor then. I will take a deeper look. 
I now use one of these: https://wiki.seeedstudio.com/Grove-Optical_Rotary_Encoder-TCUT1600X01/ with an 3D printed encoder ring, but I want to upgrade to something like: https://www.robotshop.com/nl/nl/6mm-roterende-encoder-1024-p-r.html

fluppie, 2021-08-23:
https://discourse.odriverobotics.com/t/project-hoverarm/441 Also nice lecture, this guy uses a 12-bit encoder from AMS. 65k pulses per rotation :-).

alekslnn, 2021-11-21:
Hello. It was a success to live well the result?

EFeru, 2021-11-21:
@Hylke33, sorry for late reply. It the firmware there is a possibility to input the position for the controller. Check the input structure in the bldc.c . 
Then by flipping a flag to 1, the controller will use the encoder input position. It should give better torque control even at 0 speed.

alekslnn, 2021-11-21:
Can I learn more about how to do this?

EFeru, 2021-11-21:
Here you have to plug in your encoder reading (same for the right motor):

https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/928a86b86bacf5331405fbbb990b07d82366717a/Src/bldc.c#L188

And here you set this flag to 1 to enable that your encoder input is considered by the firmware:

https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/928a86b86bacf5331405fbbb990b07d82366717a/Src/util.c#L240

I imagine a bit of calibration to get the 0 point angle of your encoder match with the 0 angle expected by the firmware. I only tested this part in simulation (not yet real life) because lack of an encoder.

alekslnn, 2021-11-21:
[image]
[image]
I use an optical 2-channel ekonder for the control board, can I use it for the horebard board? Which pins do I need to connect it to?

EFeru, 2021-11-21:
What is your sensor out outing? Voltage, UART? Then you can use ADC or UART to read the data.

alekslnn, 2021-11-21:
I am using an STm32f401 control board with EMC firmware. It gives a PWM/Dir signal -->arduino PWM Dir adapter -->UART -->Hoverboard board

alekslnn, 2021-11-21:
The encoder outputs voltage

alekslnn, 2021-11-22:
I transform the pulses into a voltage of 0-360. Where to send the signal?? to which pins?

Hylke33, 2021-11-22:
@EFeru Thanks for your reply! I did not have any time to investigate this for the last months because i'm moving between houses. I hope I have more time and space from January onward. I will post my findings then.

alekslnn, 2021-11-22:
Hylke33, Is it possible to communicate with you in a messenger to exchange experiences?

cohnhead66, 2022-04-25:
I am hoping to get this to work for a Force Feedback Wheel also.  I have an encoder set up that I can read with an arduino.  I already send torque command via PWM1, figure I will send mechanical angle via PWM2 since I am only using one wheel

I have two questions.

1 -- I assume I can process in readInputRaw() and save the angle to rtP_Left.b_angleMeasEna.  Is this where the code should go? 
#if defined(CONTROL_PWM_LEFT)
    if (inIdx == CONTROL_PWM_LEFT) {
      input1[inIdx].raw = (pwm_captured_ch1_value - 500) * 2;
      input2[inIdx].raw = (pwm_captured_ch2_value - 500) * 2;
    }
    #endif
    #if defined(CONTROL_PWM_RIGHT)
    if (inIdx == CONTROL_PWM_RIGHT) {
      input1[inIdx].raw = (pwm_captured_ch1_value - 500) * 2;
      input2[inIdx].raw = (pwm_captured_ch2_value - 500) * 2;
    }
    #endif

Question 2:  How can I calibrate between halls and encoder.  Is it as easy as finding a position where  all three sensors are low, call that 0, and then reference encoder angle from that position?  Is there a way to output  hall position?

Thanks for the help.

korvolle, 2023-07-08:
Did anyone manage to make this work?

ArtemVolkov2103, 2024-04-23:
Hello everyone, has anyone managed to use an external potentiometer or AS5600 sensor module to determine the position of the wheel instead of the hall sensor, which is inside the wheel?

adamosaurus07, 2024-11-06:
Looking forward for the progress/fix!

Mega-Rexy, 2024-11-06:
We should get a proper photo of the wirings! Some people are willing to make Direct Drive wheels out of this board!

Mega-Rexy, 2024-11-12:
Currently I'm waiting for my ST-Link V2 to arrive in some day. 
I have 600ppr rotary encoder, Arduino Leonardo already. 
I would like to know more things about setting up the board/wiring. 
So far I only know that once I upload the TORQUe firmware, everything is cool, so then we connect the PWM signal output( Arduino) to the board's "Sideboard" connection? And the encoder can stay wired to the Arduino?

ArtemVolkov2103, 2024-11-12:
> Currently I'm waiting for my ST-Link V2 to arrive in some day.
> I have 600ppr rotary encoder, Arduino Leonardo already.
> I would like to know more things about setting up the board/wiring.
> So far I [...]

It seems to me that the encoder needs to be connected directly to the Arduino, because programs on the PC should see the feedback - position. And the Arduino should be connected to the right cable from the motherboard of the gyro scooter, it can also work with a 5 volt logic signal. And yes, the firmware must be configured for PWM, not for an analog signal.

Mega-Rexy, 2024-11-12:
One of the guys from OpenFFBoard DC recommended to use Serial (USART) instead of PWM, mainly because of the low resolution on PWM use.

ArtemVolkov2103, 2024-11-12:
> One of the guys from OpenFFBoard DC recommended to use Serial (USART) instead of PWM, mainly because of the low resolution on PWM use.
> 

Using USART makes it possible to transmit a signal from -1000 to 1000, and the PWM on the arduino has only 256 values. It turns out 128 steps in the positive direction and 128 in the negative, which in the firmware code will be converted to +-1000 steps. So the transmission via USART will give greater accuracy than via PWM. 
It is also worth considering that in case of any power supply problems, the PWM signal may be distorted, and commands sent via USART are checked for correctness. This is elementary safer in the context of testing.

Notriatv, 2025-02-14:
> > We should get a proper photo of the wirings! Some people are willing to make Direct Drive wheels out of this board!
> 
> At the moment, I have managed to use an external encoder, but I set the target position of the wheel constantly, right in the firmware code. In the near future, I want to implement control via the right cable so that the Arduino can control. You can come in and see what I've changed, maybe it will help you!

Hi, do you ever use or heard abt this program:? https://github.com/ranenbg/Arduino-FFB-wheel
it sends pwm(9)+dir(11) from Leonardo. do you think it's possible to use it to control a DD hoverboard?

ArtemVolkov2103, 2025-02-14:
> Hi, do you ever use or heard abt this program:? https://github.com/ranenbg/Arduino-FFB-wheel
> it sends pwm(9)+dir(11) from Leonardo. do you think it's possible to use it to control a DD hoverboard?

Hello! No, I don't. Seems like it is software only for Arduino, and there is no any mentions about using with hoverboard. Hoverboard firmware not supposed to be a DD wheel, in torque mode it is will not give you to rotate wheel, as I know... Maybe I'm wrong. By the way, my project is for another goal)