EFeru/hoverboard-firmware-hack-FOC #211: can't reliably read pots/wheels will accelerate randomly

2by4mike, 2021-08-23:
im using RoboDurden online compiler and I have:
VARIANT_ADC
FOC_CTRL
SPD_MODE
I am using the preset current limits

So my setup is an RC receiver(Spektrum) that spins servos attached to the pots
the pots where working fine and I was able to control it fairly reliably, but then I recalibrated it and it seems to not be able to read the pots reliably anymore. when the throttle is constant the wheels will accelerate to a higher speed for ~1 second then return to the desired speed repeatedly. I know its not on the receiver/servo end because I can turn off the receiver and it will still happen. I don't know if this is because of interference, or just a way the code works. 
Also would it be worth it to replace the pots with hall effect sensors or would the interference from the motors be to great?

Candas1, 2021-08-23:
Why such a complicated setup ? Using RC variants (PWM/PPM/IBUS) wasn't ok ?
if you use the pots only manually, and calibrate it, it's not ok ?

2by4mike, 2021-08-23:
> Why such a complicated setup ? Using RC variants (PWM/PPM/IBUS) wasn't ok ?
I tried just using pwm and ppm, and with pwm it jumped around and off the stand and gave me a nice slice on my finger, and with ppm it didnt do anything.
even if I turn the pots manually it still goes to a higher speed for ~1 second semi-randomly.
when I recalibrate it does it overwrite the calibration that is already there, or could it be getting confused between 2 calibrations?

Candas1, 2021-08-23:
Wiki explains how to best use pwm or ppm.
Recalibrating overwrites the calibration.

2by4mike, 2021-08-23:
the thing is though, at least if I understand correctly, you plug the ppm port into the wire, but that would not work because I am using a Spektrum AR8000 which does not support ppm. 
I think I'm going to give pwm another try.

 I know I put the 100nf cap in-between/connecting the 2 inputs on the long wire to reduce interference, and in niklasfauth's video it looks like he uses a non-polarized ceramic cap, but I only have electrolytic caps. and after some searching I found that if you connect the negative sides of 2 caps together it acts like a non-polarized, but it looks like it has 3 inputs/outputs, so how would I connect that with the wires. 
Or could I just have 2 caps connected so the positive of one is connected to the negative of the other and vise versa, would that work as a non-polarized cap or do I need to use a proper nonpolarized cap.

If I use cat5 wire, would I be able to place the receiver further from the mainboard?

also, how necessary is the ferrite core, from what I have found it seems like it doesn't matter if it is a ring you wrap wire around, or a clip on, but apparently frequency does, so is there a specific type of ferrite core that works best or is it negligible?

Candas1, 2021-08-24:
This is one someone from the community had shared in the telegram group
[image]

2by4mike, 2021-09-06:
I got it all hooked up and somehow it works now, now its just waiting for the ferrites! Thanks!

closed 2021-09-06 by 2by4mike