EFeru/hoverboard-firmware-hack-FOC #212: The throttle and brake controls do not work smoothly

HenryNickel, 2021-08-28:
- VARIANT: HOVERCAR
- CONTROL TYPE: FOC_CTRL
- CONTROL MODE: TRQ_MODE

I'm just starting out and ask for advice from those more experienced.
I have programmed my hoverboard board with GD32F103RCT6. I used the Hovercar variant. Everything went great. I have connected the hall controller throttle and brake. I programmed everything without any problems. 
Even the calibration goes fine. But. Turning the knob slightly starts the motors running at about 50% almost all the way, and turning it to the end of the track spins the motors to full. The smooth, proportional control doesn't work at all and I don't know what the problem could be. When I apply the brake, there is a sharp twitch into reverse. The smoothness doesn't work here either. It makes the brake dangerous. I thought there was some defect in the board, but when I programmed another board, again with GD32F103RCT6, the result is exactly the same.
I tried switching the motor phases, but that was rather worse. I also have resistors at the inputs and capacitors against ground for both throttle and brake. I tried without them. Same. When I hit the brake twice, it beeps and reverse works.
Please help me. Thank you

EmanuelFeru, 2021-08-28:
Can you check that the hall input for throttle and brake are ok and noy jumping the value for some reason? You could check it with and ftdi and see the values on serial communication.
Also, you could try to redo calibration, read the wiki page for hints on how to do it correctly.
Another thing you could try is to use potentiometers just for testing purposes to check if that is smooth.

HenryNickel, 2021-08-28:
Oh great Emanuel, I tried to contact you via email first, but I guess it's not up to date anymore so I registered here. Thank you so much for getting back to me. 
I was wondering if it's a gas accellerator. That should contain a standard OH49E magnetic sensor. That would not work well with a 3.3V voltage. I tried measuring the output of the accelerator throttle with a voltmeter and the voltage varied continuously from 0.9V to 2.8V.
It's this one from China: [https://www.aliexpress.com/item/4000140693486.html?spm=a2g0o.productlist.0.0.26d66bb4tEP0yd&algo_pvid=da89ed29-5fba-4905-b179-ddcdb3a25603&algo_expid=da89ed29-5fba-4905-b179-ddcdb3a25603-40&btsid=2100bddf16038815933526029ef4df&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_](url)

HenryNickel, 2021-08-28:
I don't have the FTDI serial adapter. I have the input calibration set to 3 (AUTO) and have tried it about 20 times. Same result.

HenryNickel, 2021-08-28:
I erased the chip before programming. The first time I held the button down for 5s there was a higher tone after the beep, this is probably battery calibration etc. The next time I press it for 5s there is always a low longer tone after the beep, then I turn both controllers sequentially several times to the extreme positions. Then it stops with a chirp.

HenryNickel, 2021-08-28:
It seems to me that in reverse it reacts exactly the same way. Within 5mm of turning the handle it accelerates smoothly to 50%, the next turn the speed is the same and by the end of the turn it accelerates to 100%. When it reaches max, it bites, like it's coming to an abrupt stop or briefly starting to turn the other way. The whole hoverboard jumps and I can't hold it.

HenryNickel, 2021-08-28:
When I accelerate and then put the brake on 1/3, it brakes but the wheels wobble. Both in the same direction. The more I apply the brake, the more the wobble. A few centimetres to 10cm. When I turn the brake off and on again, they don't wobble anymore. Then I move one wheel, the other wheel turns the other way and immediately both wheels start oscillating in the same direction again.

HenryNickel, 2021-08-28:
ST-Link V2 STM32+STM8 Programmer
Firmware V2.J37.S7
[image]

HenryNickel, 2021-08-28:
My settings in txt
I didn't change anything specifically.

[Henry - settings.txt](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/files/7071014/Henry.-.settings.txt)

EmanuelFeru, 2021-08-28:
When you apply the brake it is normal to oscillate if the wheels are  of the ground. It is a characteristic of torque mode which requires load otherwise the wheel will spin to max rpm even with a small throttle input.
Anyway it should not jump. So you might need to further investigate... Could it be that you by mistake entered max speed and max torque adjustment?  ( You enter that by pressing power for 5 sec and then press again).
Also I am not sure if the YST board is compatible with the firmware. Some pin mapping might not match the YST board.

HenryNickel, 2021-08-28:
Well, the mystery is clearer now. Sergei Nikitov also told me it was normal. That led me to the mistake I was making. I found it strange that I was changing CONTROL MODE on both VLT and SPD and still the control was out of proportion. I just changed it in Control selections: #define CTRL_MOD_REQ>>VLT_MODE. Only when I have the HOVERCAR option selected, there it changes to the predefined TRQ = Torque mode.
So when I changed it in the area: VARIANT_HOVERCAR SETTINGS from the predefined #define CTRL_MOD_REQ>>>TRQ_MODE to VLT or SPD, the wheel speed changed beautifully smoothly and even the brake worked fine. The idle didn't work though - when the throttle was reduced, the revs immediately dropped quickly. So it braked automatically. In reverse, it would also go full speed. Maybe it's the FIELD_WEAK_ENA setting of 1 and FIELD_WEAK_MAX of 10. So under load in TRQ, it will work smoothly - proportionally and reliably, without any twitching?
This mode has the advantage of idle. Can't the idle be adjusted in VLT or SPD mode as well?

EmanuelFeru, 2021-08-29:
Yes, under load Trq more will work smoothly. Only Trq more has freewheeling (idle) capability. I'm other modes is not possible since they have a different working principle.

closed 2022-03-06 by EFeru

theionutzradu, 2024-12-15:
Small problem here with variant PWM, if i let only direction (channel 1) connected i can go forward and reverse, if i let only direction connected i can control direction , if i connect both i can only go backwards and left (or right).
Any ideas?