EFeru/hoverboard-firmware-hack-FOC #214: N_MOT_MAX in TRQ_MODE not work.

kris86k, 2021-08-30:
Hi.

VARIANT: ADC
CONTROL TYPE: FOC_CTRL
CONTROL MODE: TRQ_MODE
What is your application?
Lawnmower with sitting driver, three wheels with a steering wheel.

I need low speed max 7km/h. I will try to change N_MOT_MAX to really low value 10-50 but it doesn't work. Can you explain me how it work? It's any way to change max speed in TRQ_MODE?

Thanks for help.
Kris

EmanuelFeru, 2021-08-30:
For low speed I think it is better to use Speed mode. The PI controller for speed control might need some tuning to match your load.

kris86k, 2021-08-30:
Speed mode is ok for move straight. When I change direction motor gived sound like iiiii and switchd off all.

gmladeno, 2022-03-17:
you can use a low pass filter in main.c to achieve low speed  and works perfectly in speed mode , see this video  by Practical Rpobotics  it is a great job 
https://youtu.be/q451WdEPkjo

davidpiwczyks2s, 2022-06-09:
I've got the same problem - N_MOT_MAX doesn't seem to limit the speed in torque mode. N_MOT_MAX is currently set to 60, I measured the motor turning at 600RPM. I would switch to speed mode but I need the ability to freewheel.
What do you recommend?

closed 2022-07-04 by Candas1

Amitkor12, 2024-01-29:
> [image]

what exactly is flash write key and how to change it ?