EFeru/hoverboard-firmware-hack-FOC #215: Hello, teachers, I come from China, I have a question to ask

mouixong, 2021-08-30:
I in an accidental opportunity discovered that your this firmware is really too good, is extremely good, specially that 4 gears may shift gears I like it very much, but I found a problem in the process of using it, that is, it will shut down automatically after releasing the throttle, but my voltage is normal, and the battery is just full
But I do not seem to have this problem with other firmware I hope that teachers can give me pointers Teachers, do you use Wechat?You can add my wechat if you can 290970116

EmanuelFeru, 2021-08-30:
Is there a way to log the battery voltage from serial while the shutdown is happening?

mouixong, 2021-08-31:
Is there any way？

EmanuelFeru, 2021-08-31:
Serial communication using FTDI adapter

mouixong, 2021-08-31:
Is there a tutorial on this? I Don't know much about this， I have a Bluetooth module in my hand

EmanuelFeru, 2021-08-31:
With Bluetooth module is possible too. Check the wiki pages in this repo.

mouixong, 2021-08-31:
wiki above the text tutorial is relatively simple, for us this kind of rookie a little do not understand, there is this aspect of video teaching?

vamfun, 2021-09-01:
As suggested using Debug USART2 or USART3 to FTDI is easiest but it depends on how fast the shut down occurs.   The battery voltage has a low pass filter before debug print..I don’t recall the smoothing lag time constant.   But it can create a delay so you might miss the full transient before shutdown.

The best method is to use an oscilloscope to monitor voltage directly on board.   If no Oscilloscope then  try just a VOM meter but these have lags too..but at least you might see a drop if the transient isn’t too fast,

http://vamfun.wordpress.com
Robot musing blog
iPhone

Candas1, 2022-01-02:
[Wiki](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Battery/_edit#troubleshooting) has now more details to troubleshoot such issues.

closed 2022-01-02 by Candas1