EFeru/hoverboard-firmware-hack-FOC #216: Transpotter: Dual input with VARIANT_ADC via transpotter Line and RC Receiver

derFrickler, 2021-09-01:
### Your System
- VARIANT: VARIANT_TRANSPOTTER
- CONTROL TYPE:  VARIANT_ADC via transpotter Line

- What is your application?
    - Transpotter platform but would like to add RC control aswell.

Thanks a lot for your great firmware! After struggling a lot with other firmwares 1-2 years ago because my board was not really supported i just flashed hoverboard-firmware-hack-FOC with board variant set to 1 and it seems to work great!

Just one Question, would it be possible to add dual input to the Transpotter as well? 
Meaning if the Transpotter Line is not pulled out it would be controllable via RC PPM/PWM?

Would be great if you could give some hint on how to archive this.

Thanks a lot
      Johannes

Candas1, 2021-09-03:
Hi,

Unfortunately Transpotter variant is using separate code, so it's not using the latest features like calibration, dual inputs,...
I once started to analyzed how to streamline but never had the time to progress.

Regards,

Candas

fivef, 2023-02-04:
Before I start to implement something I wanted to ask if there are any updates on this? @derFrickler did you succeed?

derFrickler, 2023-02-04:
Hi, no I actually never tried. I just changed the firmware whether I wanted transporter or remote