EFeru/hoverboard-firmware-hack-FOC #217: Using only one wheel of the hoverboard

gicutabacaru, 2021-09-08:
Hello, I would like to controll only one motor, from the motherboard side, I managed to flash with the ADC VARIANT, but it is not moving the motor. Also is making a continously beep, like it does not see the second motor, which is right. I have put two potentiometers, instead of the sideboard, but when I turn one of them I can hear a pitch sound, and this is all.
  Is there any modification to do? I didn't find any other post where someone is trying to control only one motor, when the other is missing. Thank you

Candas1, 2021-09-08:
Hi,

You need to disable one wheel
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/928a86b86bacf5331405fbbb990b07d82366717a/Inc/config.h#L145

gicutabacaru, 2021-09-08:
> Hi,
>
> You need to disable one wheel
> https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/
> 928a86b86bacf5331405fbbb990b07d82366717a/Inc/config.h#L145

Thanks a lot, it worked after disabling the left motor and calibration of
the pot.
Best regards!

closed 2021-09-08 by Candas1