EFeru/hoverboard-firmware-hack-FOC #218: Support for different hardware (SMART-L-V2.1, two controller boards equipped with STM32F103C8T6)

andishock, 2021-09-08:
Hello folks,
I know this firmware supports only the hardware consisting of the single mainboard controlling both of the hoverboard motors. I just wonder, if anybody around here stumbled over a different kind of hardware consisting of two identical controller boards each equipped with a STM32F103C8T6. There is a printing "SMART-L-V2.1" onto it. The pins for the JTAG/SWD connectors I soldered myself, so don't worry about that.
Does anybody know about some custom firmware supporting this kind of hardware?
Thank you!

[image]

Candas1, 2021-09-21:
Hi,

Have you checked [this](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility)

andishock, 2021-09-24:
Hi,
Thanks for our answer!
Yes, I already checked for similar boards out there, but none of them seems to be compatible to the hardware I found.

Candas1, 2021-11-16:
Actually in the meantime I found out the split board firmware is not working on STM32F103C8T6, someone would need to port it.

Candas1, 2021-12-27:
This new port might be the right place to start:
https://github.com/JohMah/Hoverboard-JC2015.7.31.V2.0-Hack

The board looks different so pin mapping would have to be checked, but at least the chip looks compatible.

closed 2021-12-27 by Candas1

L1mp4, 2023-07-20:
Hi! Did you find anything simple to make this run like you want it? Thanks!

zimmii91, 2024-05-05:
hi ! i got the same board, in 2 hovers actually. any luck on programming?

I-hate-2FA, 2024-05-05:
so far no gen2 f103 support on any firmware yet other then the ones modified from gen1, but i am working on porting mine to f103

Candas1, 2024-05-05:
> This new port might be the right place to start:
> https://github.com/JohMah/Hoverboard-JC2015.7.31.V2.0-Hack
> 
> The board looks different so pin mapping would have to be checked, but at least the chip looks compatible.

Hum hum