EFeru/hoverboard-firmware-hack-FOC #220: 4 wheel transpotter

softwarecrash, 2021-09-12:
i will build a transpotter, but my terain is not plane, so i think about 4 wheels transpotter that move like a tank, it is possibil to configure the firmware for that? controlled by the classic line with poti and optional with nunchuk

Candas1, 2021-09-21:
With transpotter variant there is not much room for configuration, it is not benefitting some of the last enhancements like dual inputs that could be used to also control it with a nunchuk.
Probably you need to try 2 hoverboards with transpotter variant and single gametrak as input, and see how it behaves.

lsls5921, 2022-07-16:
Is there any way to control the DC brush motor on the hovering board