EFeru/hoverboard-firmware-hack-FOC #222: Unable to connect thought USART

Dario1989a, 2021-09-16:
### Your System
- VARIANT:VARIANT_USART
- CONTROL TYPE: No one for now
- What is your application?
    - Right now simply nothing
  
Hello everybody!
I'm experiencing an issue with the bord, I cant dialogue through the USART connection with the Raspberry PI 4 GPIO USART PIN.
I've already disconnected the blueetooth and enabled the UART serial GPIO pin (Pin 8 and 10 on RPI4). I've already tried a loopback test connecting those two pins with minicom at a bauderate of 9600 and i works, so I know the device is right (`/dev/ttyAMA0` on RPI4) and the pins are correct.

Then thought Platform.io I've build the USART variable and successfully flashed the firmware.
I've got the three beeps that, according to #21  means that  is a Serial Timeout so i have to gave a Start frame and checksum.
So I've connected the RPI to the mainboard and tried this python script (that i found on issue #78 ), that should do the Start Frame and Checksum (I've just modified the serial port and the baudrate):

```
import time
import serial
import struct
import ctypes

def check_num(num):  # if number is negative convert it and put it in integer variable, else don't change it
    if num < 0:
        num_out = ctypes.c_ushort(num).value
    else:
        num_out = num
    return num_out

def pack(steer, speed):  # pack to package
    start_frame = 43981
    u_steer = check_num(steer)  # check and replace if number is negative
    u_speed = check_num(speed)  # check and replace if number is negative
    # start frame, steer, speed, checksum
    package = struct.pack("<HHHH", start_frame, u_steer, u_speed, (start_frame ^ u_steer ^ u_speed))  # H-unsigned short
    # print("pack: ", package)
    return package

ser = serial.Serial(port='**/dev/ttyAMA0**', baudrate=**115200**, timeout=1)  # define serial (set your port!!!)

# prepared package for stop, first \xcd\xab --> start frame 0xABCD and last \xcd\xab --> checksum
stop = b'\xcd\xab\x00\x00\x00\x00\xcd\xab'
cnt = 0

while True:
    if cnt < 50:
        # to work first command must be that:
        ser.write(stop)  # stand still
        ser.write(0)  # need to send to work
    elif cnt < 200:
        ser.write(pack(0, -50))  # drive backward
        ser.write(0)  # need to send to work
    elif cnt < 300:
        ser.write(pack(0, 50))  # drive forward
        ser.write(0)  # need to send to work
    elif cnt < 400:
        ser.write(pack(100, 50))  # steer
        ser.write(0)  # need to send to work
    elif cnt < 500:
        ser.write(pack(100, 10))  # steer
        ser.write(0)  # need to send to work
    elif cnt < 550:
        ser.write(stop)  # stand still
        ser.write(0)  # need to send to work
    else:
        break
    print(cnt)
    cnt += 1
    time.sleep(0.05)  # main loop interval 50ms

print("End")
```
Tried to connect to minicom for sending the instruction in Hex, tryed `tail -f /dev/ttyAMA` listening for some response from the hoverboard but nothing.
Some of you have any suggestion on that?
My cable configuration is this one:
[image]

[image]

PS: already tried to switch the TX RX cable on RPI but doesn't work, also measured the voltage on the Right sensor board to be sure that ground is ground and the 15V Pin is leaved disconnected.

Candas1, 2021-09-21:
Hi,

You board looks a bit different than the usual ones, have you checked if the [pin mapping](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility#pin-mapping) is the same ?

Dario1989a, 2021-09-22:
Hi! Thanks for you response.
Actually I think i have a really strange board. I've struggled two day figuring out why the firmware wont flash and I've found that the programmer's pin are in different order, so I think this board has a different pin mapping. 
May I ask a question?
I'm trying to find the RX and TX pins thought a continuity test with a probe shifting on the whole pin on MCU and the other one on the left sensor board but I was only able to find the GND connection (for example i found that the SWD, SWCLK pin are on others pin than the reverse schematic that i found).
Because I'm pretty new to electronics, do you think that this continuity test with the battery unplugged will help me to find those pins?
According to STM32F103RCT6 datasheet these should be my pin configuration
[image]
[image]
?
[image]

Candas1, 2021-09-22:
As per the wiki
-TX should be connected to PA2 and RX to PA3 if you use USART2 (left sensor cable)
-TX should be connected to PB10 and RX to PB11 if you use USART3 (right sensor cable)

With continuity check you should be able to trace it back to the MCU.

Dario1989a, 2021-09-22:
I've tried so far but seems like, apart from ground and Vcc I'm not able to find the TX and RX pin. I've tried the left sensor board, right sensor board, and hall cable too. Apart from ground no one beeps. 
Maybe the board is just broken, but it's seems strange to me that all the MCU-pin connection are burned. 
The fact that the board is disconnected from battery should help not to burn the multimeter and shouldn't affect the measurement.
I've tried to measure the resistence between PB11 and i found 0.9 Ohm (seems a quasi continuity but multimeter won't beep).
The measurement between PB10 and the other pin gave me 100 Ohm(no beep but should be ok no?).

Candas1, 2021-09-23:
There are other components between the MCU and the sensor cable, you can skip those components and check the continuity closer to the MCU.

Candas1, 2021-09-23:
Maybe your board is just ok, you can also try to control it with USART from a different device ?
Like a FTDI and the web tool mentioned in the wiki ?

Dario1989a, 2021-09-23:
Yeah I've just put a probe in each terminal of the right sideboard and the other one going throught all the pins of MCU. The strange thing is that doesn't detect even the left sensor and the hall (searching through all the MCU pins). I've tried with a FT232 GND, supposed TX and RX, supposed RX to TX (and the opposite) with your WEB serial connecting the right USB device and changing the Baudrate-> 3 Beeps and no output on the console

Candas1, 2021-09-25:
You have to select receive = binary to see the data in the console with usart variant. And protocol = usart in Ctrl view so that commands are sent and firmware is not beeping

Candas1, 2022-01-02:
Hi, is it working now ? can this issue be closed ?
[Wiki](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-USART) has now more details that can help.

Dario1989a, 2022-01-03:
Thank you Candas, I bought a new board and throught your fantastic Webserial Debug worked!

Candas1, 2022-01-03:
Ok so it was definitely a defective board.
Thanks for your feedback !!!

closed 2022-01-03 by Candas1