EFeru/hoverboard-firmware-hack-FOC #223: Turn Mosfets off on start or throttle < 10

valera3258, 2021-09-18:
Hi there, is there any solution to get transistors off when throttle is close to zero or when i turning on device, like on hoveboard to get free wheel in ADC, FOC_CTRL?
As well, when wheel is spinning faster than throttle is pressed, make mosfets turned off or not interfere to wheel control, only listening inputs.
i'm tryed all control modes in FOC_CTRL, and no one of them looks safe for bike with hover motor.
Thanks.

Candas1, 2021-09-21:
Hi,

Is it not what Torque mode is for ? In this mode 0 command means freewheeling.
Have you tried it ?

valera3258, 2021-09-26:
Yes im tried, and its not wat i expected for.
The wheel tries to immediately change speed, and this is a problem, in other side, i can't get trottle value smooth like 0-100 (0.8v-2.535v) in this mode.
Im tried to change code, but no results, needs more practice with coding for stm32, im more on c# or java, so therefore i asked.

Candas1, 2021-09-27:
I don't understand what you mean with your first sentence.

For the non linearity in torque mode, assuming you calibrated well your inputs, it's a known [issue](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/issues/167) discussed here.

valera3258, 2021-09-28:
I have bike with 16 inch wheels, and mounted in front wheel 350w motor, if kid riding from hill on 25km and tries to press throttle pedal on 5% wheel starts stopping, converting energy in heating, and im try to fix this in code, sorry if i cant explain correct.

Candas1, 2022-01-02:
Did you finally get it to work ?

Jenya-m, 2022-02-22:
Most likely, your halls are not connected correctly ... (I have come across this. The wheel seems to spin smoothly, but it is buggy when braking)
// #define MOTOR_LEFT_ENA
#define MOTOR_RIGHT_ENA
#define CTRL_TYP_SEL FOC_CTRL
#define CTRL_MOD_REQ TRQ_MODE
#define FIELD_WEAK_ENA 0
#define VARIANT_HOVERCAR
#define CTRL_MOD_REQ TRQ_MODE
works for me so great.

closed 2022-05-11 by Candas1