EFeru/hoverboard-firmware-hack-FOC #224: Motors spinning at max speed

AleLinox, 2021-09-20:
### Your System
- VARIANT: USART 
- CONTROL MODE: SPD_MODE
- What is your application? Four wheeled robot

Hi everyone, I'm currently trying to flash Emanuel's firmware on the board in the picture. The main control is an GD32F103, compatible with the firmware and in fact I'm normally using this on another board. But on this one when I turn on the supply the motors start to spin to max velocity 
[image]
and I don't understand way. Can someone help me?

Candas1, 2021-09-20:
I think @EmanuelFeru once mentioned this can happen in speed mode when hall sensors are not mapped in the way expected by the firmware. So the firmware reads a speed with an opposite sign, and increases the torque to reach commanded speed, but will never reach it and limits at max rpm. You probably need to find the right hall sensor mapping.

AleLinox, 2021-09-21:
Yeah, after erasing multiple times and changing the hall sensors position now all works fine. Thanks @Candas1

closed 2021-09-21 by AleLinox