EFeru/hoverboard-firmware-hack-FOC #225: UART

martin2021-sh, 2021-09-22:
### Your System
- VARIANT:
- CONTROL TYPE:
- CONTROL MODE:
- What is your application?
    - I use the arduino board to control the controlling board of MOTOR via Hardserial serial1. there is noise from one motor tyre to another motor type. It seem the motor is braked. I don't  know what is wrong with it . Is the issue from firmware. pls help me to figure out. If it is solved,  I will give you a little present.

**Before asking a question** see if it is already answered in:
* [Wiki pages](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki)

**Indicate what you have tried:**
If applicable, indicate what you tried that _doesn't_ work:

```c
input1[inIdx].raw = adc_buffer.l_rx2;
input2[inIdx].raw = adc_buffer.l_tx2
```

Candas1, 2021-09-22:
Do you have a video to share ?

martin2021-sh, 2021-09-25:
Hello EmanuelFeru.
THX for your reply . I have already solved the issue. I  wrongly connect the hall wire. Keep in touch.
Best Regards,
                                                             Martin shen
                                                         25/09/2021

closed 2021-09-25 by EmanuelFeru