EFeru/hoverboard-firmware-hack-FOC #226: Brake work the wrong direction

Flash468, 2021-10-01:
### Your System
- VARIANT: HOVERCAR
- CONTROL TYPE: ADC
- CONTROL MODE: TRQ_MODE
- What is your application?
    - Hovercar

Hello, i have flashed the board and the sideboard with the HOVERCAR configuration and ajusted the two brake and accelerate pedals with the auto calibration. I don´t have changed anything in the Hovercar configuration. It is completely standard. The accelerate pedal work fine but the brake pedal work the wrong direktion. When the wheels are spinning and i trigger the brake the car accelerate und don´t decrese the speed. The same when the car drive reverse. So i thing the direction of the brake should be inverted. I don´t found any configurations to ajust this. What do i have to change to make the brake work?

Thanks

Candas1, 2021-10-14:
Hi,

I think this can also happen if the phases/hall are not wired as expected.
You can try other combinations.

Candas1, 2021-10-16:
I tried swapping blue and green wire on both phases and halls, and I was also getting a weird behavior because of wrong speed sign.

You can try different combinations.

Flash468, 2021-10-17:
Today i have tested every single combination without success.
I have changed the pining from the hall sensors in defines.h and flashed it every time. And with each of this changed pinings i have tested all possible motor phase combinations.
#define RIGHT_HALL_U_PIN GPIO_PIN_10, 10, 11, 11, 12, 12
#define RIGHT_HALL_V_PIN GPIO_PIN_11, 12, 10, 12, 10, 11
#define RIGHT_HALL_W_PIN GPIO_PIN_12, 11, 12, 10, 11, 10
with each
blue:      blue,     blue,     yellow,  green,  yellow, green
green:   green,   yellow,  green,   blue,    blue,    yellow
yellow:  yellow,  green,   blue,     yellow, green,  blue

I had 3 possible results
- No spinning of the wheel
- rough spinning, the brake work as it should
- smooth spinning, the brake accelerate instead of braking

closed 2022-05-11 by Candas1