EFeru/hoverboard-firmware-hack-FOC #228: Rotation noice from motors when under strain

frederik1012, 2021-10-04:
### Your System
- VARIANT: hovercar
- CONTROL TYPE: foc, sin and com
- CONTROL MODE: volt mode
- What is your application?
    - Hovercar

I have a strange noise coming from both motors when they are under medium to high load. The behavior only occurs then field weakening is enabled. The torque from the motors still remain, but the noise is quite loud. The sound is like something is rotating inside the motor, but no mechanical part is moving, except the motor of course. And now I can’t seem to turn off field weakening. 
```c
// a code sample may improve communication
```

Candas1, 2021-11-16:
Did it work ?

frederik1012, 2021-11-23:
Definitely less noise. and more strain is required for the motors to start the noise. I think the problem is a combination of a not so powerful battery, and a larger wheel diameter than normal. With normal wheel diameter the problem is gone.

Candas1, 2021-11-23:
I am wondering if it can also due to current chopping due to reaching max current being too low for your use case.
It can be increased in config.h but you shouldn't increase too much.

frederik1012, 2021-12-09:
Unfortunately I can’t test this as my project has taken a turn in another direction, but I appreciate all of the help that you are giving, and hope this thread can be of any help to others

closed 2021-12-09 by frederik1012