EFeru/hoverboard-firmware-hack-FOC #236: hello,not a firmware issue,but mcu combatiblity

miso521, 2021-11-05:
### Your System
- VARIANT:pwm
- CONTROL TYPE:hoverboard
- CONTROL MODE:torque

Hello
soo im trying to upload this code to stm32f103c8t6,cuz thats what i have for now,but as i cann see this is for STM32F103RCT.i tied to complie it for my mcu but i gives me libraries errors,soo im wondering could this be complied for my mcu.

thank you.

Candas1, 2021-11-05:
Hi, have you checked this ?
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility
How is your board looking like ?

miso521, 2021-11-08:
hi.
i have board with STM32F130C8 chip,firmware you linked,as far i cann see its not supporting pwm dir input as yours does.

Candas1, 2021-11-08:
ok so what you initially wrote was wrong.
The firmware from Emanuel will not work on split boards.

Maybe check the other forks of the split board firmware, someone might have extended it.

closed 2021-11-21 by EFeru