EFeru/hoverboard-firmware-hack-FOC #237: compatibility question - Unique mainboard and 6s battery

j8d, 2021-11-08:
### Your System
- VARIANT: ADC
- CONTROL TYPE: FOC
- CONTROL MODE: TRQ
- What is your application?
    - eTrike

Curious if anyone has come across the following board? Trying to determine if a) I'm going to brick it attempting to flash and b) if the board can handle a 10s (or maybe 12s) battery. It has GD32F103RCT6 and the two 200u input capacitors are rated at 50v. 

[image]

Candas1, 2021-11-08:
Hi,

It looks very different.
I wouldn't flash it without checking the pins are mapped in the same way:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility#pin-mapping

j8d, 2021-11-09:
Pinout appears to be standard. Going to flash it and hope for the best. 

Also, here's a visual of pinout I put together in case anyone else finds it useful. 

[image]

j8d, 2021-11-13:
Stuck in bootloop after flash. Power on chime, low tone beep, power off chime. Anything I can do or is it bricked?

Candas1, 2021-11-13:
Tick all the option bytes in st-link utility

j8d, 2021-11-14:
Thanks! I was able to erase the chip but I'm still getting a bootloop when I flash the firmware. Going to try some other builds.

Candas1, 2021-11-14:
But have you ticked WDG_SW option?

j8d, 2021-11-15:
Yes, I got it to stop powering off by reducing BAT_CALIB_ADC. But no motor control via pots, and I'm no longer able to power down the board without disconnecting the battery. Suspect something with my having a 24v battery?

Candas1, 2021-11-15:
Have you checked this?
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Battery

j8d, 2021-11-16:
Thanks Candas. Will work on getting serial debug working

Candas1, 2021-11-16:
I had overlooked that you were using a 7s battery.
What you describe is not a loop, the firmware is turning of because of the battery voltage being too low for 10s.
Some users really experience a loop, with the firmware turning off/on/off/on, but this is related to the option bytes/watchdog.

The link describes how to setup the right number of battery cells, this will solve your problem.
The calibration is not mandatory but it helps have a more accurate voltage measurement.

j8d, 2021-11-16:
Thanks. Yes, I already had BAT_CELLS set to 7 and BAT_CALIB_REAL_VOLTAGE set to measured value (2529). Hopefully debug will provide more info. Maybe a hardware issue. Worst case I have a second board I can try flashing

j8d, 2021-11-21:
Unable to get either board working with the firmware. These appear to be the standard main boards in Swagtron brand 24v hoverboards. Happy to donate these two boards to anyone interested in exploring further.

Candas1, 2021-11-21:
It's still powering off ? With chime? 
There can be only 2 reasons for the firmware to power off, low battery or inactivity. You can try to disable the power off, would give you time to check the debug output

j8d, 2021-11-29:
I just finished flashing a standard st32F103 36v mainboard. Worked great before flash. Now, momentary press of the power button does nothing. Holding power button for >2 seconds causes board LED to illuminate until released. But the board won't power on and no serial debug data. Any ideas? The battery is older and will only charge to 36.18v. Could that be the issue?

Candas1, 2021-11-29:
If there is no power on tune, the chip is either not flashed properly or defective

Candas1, 2022-01-02:
Hi, I believe you were finally successful, was it with the same board ? this issue can be closed ?

closed 2022-03-06 by EFeru

SriharshaShesham, 2024-06-01:
> ### Your System
> * VARIANT: ADC
> * CONTROL TYPE: FOC
> * CONTROL MODE: TRQ
> * What is your application?
>   
>   * eTrike
> 
> Curious if anyone has come across the following board? Trying to determine if a) I'm going to brick it attempting to flash and b) if the board can handle a 10s (or maybe 12s) battery. It has GD32F103RCT6 and the two 220u input capacitors are rated at 50v.
> 
> [image]

Hi,
I have the exact same board. Could someone please provide the pin connection diagram. I keep getting device not found error.

WachtelM, 2025-05-23:
I know that this thread is very long ago. But I just tried around with the board and while looking carefully at it I noticed that the programming pin is exactly upside down from all the usual boards. If you flip your programming wires it should work