EFeru/hoverboard-firmware-hack-FOC #239: Any Chinese link where to buy a compatible PCB?

tomy983, 2021-11-11:
As the title says...
Thank you

Candas1, 2021-11-15:
Hi,

There are really few boards in aliexpress that look compatible, and it costs 40-50e, it was costing around 25e in the past.
There are cheaper black boards but to my knowledge it's not [compatible](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Firmware-Compatibility) with this firmware.

Many in the community buy used hoverboards, but there is always a risk that the board is not compatible (e.g. split board).

A German member from the [telegram group](https://t.me/joinchat/BHWO_RKu2LT5ZxEkvUB8uw) is selling some new compatible boards if you are interested.

tomy983, 2021-11-19:
Thank you Candas, I'll have a look!

closed 2021-12-05 by tomy983