EFeru/hoverboard-firmware-hack-FOC #242: dual inputs, USART and IBUS

Tjaart-Hobby, 2021-11-24:
### Your System
- VARIANT: attempting dual input USART and IBUS
- CONTROL TYPE: using FOC
- CONTROL MODE: speed
- What is your application?
    - Robot usart(ROS1 later ROS2) with RC IBUS override

**Before asking a question** see if it is already answered in:
been at it for days, the dual part of it

**Indicate what you have tried:**
So got the hardware, can flash the board, have USART working, have IBUS working (small adjustments needed) can use Arduino uno to sent serial steer and speed values.

Can't get dual inputs working. It says "minor adjustments in gonfig.h can achieve dual IBUS and USART. "

Please any help.

```c
input1[inIdx].raw = adc_buffer.l_rx2;
input2[inIdx].raw = adc_buffer.l_tx2
```

EFeru, 2021-11-29:
To get UART (prority 0) with IBUS (priority 1), you can use **VARIANT_IBUS**:

[image]

Then, follow these steps:
1. Enable DUAL_INPUTS: uncomment this line:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/928a86b86bacf5331405fbbb990b07d82366717a/Inc/config.h#L464
2. Replace ADC_CONTROL with Serial control on the left cable:
- delete this line
https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/928a86b86bacf5331405fbbb990b07d82366717a/Inc/config.h#L467
- and add the lines below in the same location:
```
#define CONTROL_SERIAL_USART2  0
#define FEEDBACK_SERIAL_USART2
``` 
3. Update the min/max limits here:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/928a86b86bacf5331405fbbb990b07d82366717a/Inc/config.h#L470-L471
with:
```
 #define PRI_INPUT1            3, -1000, 0, 1000, 0  // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
 #define PRI_INPUT2            3, -1000, 0, 1000, 0  // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
```

Tjaart-Hobby, 2021-11-29:
You are a legend. Thanks  will test it Sat 4. (Work away from home)

Tjaart-Hobby, 2021-12-04:
Hi, home at last
Still not working though.
>Have IA6B receiver working on USART3
>Have Arduino serial working on USART2 (connected through logic converter)
Does not work when using VARIANT_IBUS defined inplatformio ini with suggested changes.

`// ################################# VARIANT_IBUS SETTINGS ##############################
#ifdef VARIANT_IBUS
/* CONTROL VIA RC REMOTE WITH FLYSKY IBUS PROTOCOL 
* Connected to Right sensor board cable. Channel 1: steering, Channel 2: speed.
*/
  #define CONTROL_IBUS            0       // use IBUS as input. Number indicates priority for dual-input.
  #define IBUS_NUM_CHANNELS       14      // total number of IBUS channels to receive, even if they are not used.
  #define IBUS_LENGTH             0x20
  #define IBUS_COMMAND            0x40
  #define USART3_BAUD             115200

  #define DUAL_INPUTS                     // ADC*(Primary) + iBUS(Auxiliary). Uncomment this to use Dual-inputs
  #ifdef DUAL_INPUTS
    #define FLASH_WRITE_KEY       0x1106  // Flash memory writing key. Change this key to ignore the input calibrations from the flash memory and use the ones in config.h
    //#define ADC_CONTROL           0       // use ADC as input. Number indicates priority for dual-input. Disable CONTROL_SERIAL_USART2, FEEDBACK_SERIAL_USART2, DEBUG_SERIAL_USART2!
    #define CONTROL_SERIAL_USART2  0
    #define FEEDBACK_SERIAL_USART2
    #define CONTROL_SERIAL_USART3 1       // use RC iBUS input on the RIGHT cable. Number indicates priority for dual-input. Disable DEBUG_SERIAL_USART3!
    #define FEEDBACK_SERIAL_USART3        // right sensor board cable, disable if ADC or PPM is used!
    #define PRI_INPUT1            3, -1000, 0, 1000, 0  // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
    #define PRI_INPUT2            3, -1000, 0, 1000, 0  // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
    #define AUX_INPUT1            3, -1000, 0, 1000, 0  // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
    #define AUX_INPUT2            3, -1000, 0, 1000, 0  // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
  #else
    #define FLASH_WRITE_KEY       0x1006  // Flash memory writing key. Change this key to ignore the input calibrations from the flash memory and use the ones in config.h
    #define CONTROL_SERIAL_USART3 0       // use RC iBUS input on the RIGHT cable, disable if ADC or PPM is used!
    #define FEEDBACK_SERIAL_USART3        // right sensor board cable, disable if ADC or PPM is used!
    #define PRI_INPUT1            3, -1000, 0, 1000, 0  // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
    #define PRI_INPUT2            3, -1000, 0, 1000, 0  // TYPE, MIN, MID, MAX, DEADBAND. See INPUT FORMAT section
  #endif

  #if defined(CONTROL_SERIAL_USART3) && !defined(DUAL_INPUTS)
    #define DEBUG_SERIAL_USART2           // left sensor cable debug
  #elif defined(DEBUG_SERIAL_USART2) && !defined(DUAL_INPUTS)
    #define DEBUG_SERIAL_USART3           // right sensor cable debug
  #endif
#endif
// ############################# END OF VARIANT_IBUS SETTINGS ############################`

also tried changing #define CONTROL_IBUS  to 1 
I am testing on board version 1 (the one with the white wire from motor)

EFeru, 2021-12-04:
Does it work in single mode (meaning using one input only)?
Check that it works without dual input, one time for ibus and separate for UART.
If both are ok, the dual input should normally also work.

Tjaart-Hobby, 2021-12-04:
jip it works in single mode for both the rc receiver on right/short cable and after re-flash it works for USART on left/long cable through bi-directional logic level shifter.  but not for both

EFeru, 2021-12-04:
Any beeps or something?

Tjaart-Hobby, 2021-12-04:
yes, steady beep. like when no USART input detected, I will up load a vid to youtube. Thank you again for the help

Tjaart-Hobby, 2021-12-04:
not good video, hard to do with only two hands, sorry.
1st flashed uart on left cable via logic level shifter - works
2de flashed RC on right cable -works
3de tried dual but only received steady beep, like it is not detecting signal
link to video on youtube [https://youtu.be/XPpBaZ_8puk](url)
HD upload still in progress but SD should be available

Tjaart-Hobby, 2021-12-06:
seems link does not works I will try again. [https://www.youtube.com/watch?v=XPpBaZ_8puk](url)

Candas1, 2021-12-06:
Hi @EFeru, I could only have a look to this issue now.
I think the way IBUS was coded, firmware now expects IBUS on both inputs as we don't differentiate IBUS_LEFT or IBUS_RIGHT.

https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/cf7b1d0de1da34612b056c316480f47702a15cbe/Src/util.c#L1224

One possible workaround for Tjaart would be to use SIDEBOARD_USART instead of CONTROL_USART, but he would need to adapt the arduino code to send SIDEBOARD frames.

EFeru, 2021-12-14:
@Tjaart-Hobby I checked, and yes I confirm what @Candas1 said, because IBUS is a subset of SERIAL_CONTROL at the moment with the above settings you get dual IBUS variant on both cables, so basically the firmware expects IBUS input on both Left and Right cable. 

To separate IBUS on Right and Serial on Left for example we need to add CONTROL_IBUS_LEFT and CONTROL_IBUS_RIGHT and manage the code accordigly. It is possible but requires some work.

Is it a blocker for you? Because I see 2 workarounds:
1. You mimic the IBUS protocol on the Arduino side (it is not difficult), you just need to send the data in the following structure:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/cf7b1d0de1da34612b056c316480f47702a15cbe/Inc/util.h#L30-L36
For sure online there is info on how to fill in the structure. You can also check in this firmware and do the reverse. @Candas1 also did it in his monitoring tool.

2. You can maybe use PWM + Serial control? I know PWM is not ideal but maybe is sufficient for what you need.

Tjaart-Hobby, 2021-12-16:
Hi EFeru and Candas

Apologies for the late reply. Mentioned I work away from home.
Suggested work around 1 : You mimic the IBUS protocol on the Arduino side>
The arduino is only used as a temporary USART test. It won't be part of the final robot
[image]
The tall one topples over when stopping or accelerating to quick. So since last I have been trying to use USART (right cable) and sideboard (left cable). The RC receiver connected to sideboard. That way I can later use the MPU to temper the brake/acceleration as to not topple over as much.
I have not considdered The PWM + Serial control yet, I will have a look.

If I could get the sideboard with RC receiver working it would be a work around for both. Not ideal to have a "unnecessary" extra board in the low profile one but I really want to get on with operating them through ROS. (Robot Operating System)[https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwia1dWFi-f0AhXTzTgGHWCXD1AQFnoECAUQAQ&url=https%3A%2F%2Fwww.ros.org%2F&usg=AOvVaw0fK_jn29t6rX-LcJJc2Sju](url)
Kind Regards

EFeru, 2021-12-16:
Sideboard with IBUS to Rc and mainboard with both cables Serial is indeed best if you want to also use the IMU on the sideboard.

Tjaart-Hobby, 2021-12-16:
> Sideboard with IBUS to Rc and mainboard with both cables Serial is indeed best if you want to also use the IMU on the sideboard.

I am having dramas with the sideboard code but will keep at it. Thanks for the help.
If you could some day code a variant for setups like these robots you would go from hero to superman :)
Thanks again

Tjaart-Hobby, 2022-01-19:
So in the end and a better option in any way is to use USART variant with sideboard serial as dual input.
[https://github.com/EFeru/hoverboard-sideboard-hack-STM/issues/8](url)

closed 2022-01-19 by Tjaart-Hobby