EFeru/hoverboard-firmware-hack-FOC #244: PWM speed channel not working

villamany, 2021-12-13:
Hi everyone. New on this project.
I'm testing the PWM variant but while the steering works the speed dind't it. Tested the ADC variant and works as expected with two pots. 
In the PWM variant the "in2" always read 0 when serial debug. The output train of the RC receiver is ok on both channels.

Thanks

Candas1, 2021-12-13:
Have you wrongly done the auto calibration maybe?
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Input-Calibration#auto-calibration

You can check debug output when executing this procedure.

villamany, 2021-12-13:
yes, but also seems to read always 0.

Input calibration started...
Input1 is a mid-resting pot..OK
Input2 is ignored..OK
Limits Input1: TYP:2 MIN:-999 MID:2 MAX:998
Limits Input2: TYP:0 MIN:0 MID:0 MAX:0
-- Motors enabled --

Candas1, 2021-12-13:
The firmware works for sure, I have tested pwm myself.
Calibration doesn't seem to detect anything. 
So there could be an issue with your receiver, wiring, or mainboard.

You are saying ADC  works but ADC variant is using left cable, and PWM variant is using right cable by default (shorter). Have you tried pwm on left cable to make sure you don't have a problem on the right cable?

villamany, 2021-12-13:
Yes, tested on both right and left cable by "#define CONTROL_PWM_LEFT/RIGHT",  same result.
Receiver CHs are ok (tested with oscilloscope), and if i swap the CH1-CH2 wires steering works with CH2 instead CH1, wires also are ok.

I don´t know why the debug shows alway Input2 as TYP:0, although I defined it as type 2

Candas1, 2021-12-13:
You don't have to change the type.
It will become 0 if you do the auto calibration and  no input is detected.
To be on the safer side erase the chip, flash the firmware, and try without entering the Auto-calibration

villamany, 2021-12-13:
Same result, full chip erase, redownload the sources, set to PWM mode, and upload. 
Input 2 still reads always 0.

villamany, 2021-12-13:
[image]

[image]

Candas1, 2021-12-13:
The board looks similar.
When you say that you see a signal for CH2 on the oscilloscope, you see the duty cycle changing when moving the throttle ?

villamany, 2021-12-13:
> The board looks similar. When you say that you see a signal for CH2 on the oscilloscope, you see the duty cycle changing when moving the throttle ?

Yes

Candas1, 2021-12-13:
The code hasn't changed for a while, and other community members shared videos of their setup with pwm.
But would be weird that's it's not working on both sides.
Is you receiver only handling PWM ?

villamany, 2021-12-13:
Tested the PPM variant with a PPM receiver (used the IN2 wire acording to wiki -green wire-), no any signal detected, always shows IN1=0 and IN2=0, so i assume a faulty board in both left/right ports with the same issue!?

Candas1, 2021-12-13:
Are you sure you use the right number of channels?
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-PPM

villamany, 2021-12-13:
In PPM i tried first with16 channels and second select only 4 channels in the Taranis TX configuration. 
Of couse im using Ail and Elev sticks remaped to CH1 and CH2

Candas1, 2021-12-14:
No beeps ? You should get beeps if the number of channel is incorrect.
Same on left and right ?
What receiver are you using ?

villamany, 2021-12-14:
I got 4 low beps (no PPM input), i matched the PPM CHs with my PPM RC receiver by  #define PPM_NUM_CHANNELS        8
Checked with oscilloscope my RC PPM train, all seems ok, but still no inputs (the two inputs reads 0)

Here is my PPM signal with the CH1 at minimun (1000ms) and the CH2 at maximun (2000ms), CH3-8 are not used (1500ms)
[image]

Candas1, 2021-12-14:
So even after matching the channels, you get beeps, and from both sides ?

villamany, 2021-12-14:
yes, tried both sides and both wires on each side.
At power On i got first two low beeps (no ADC) and then start to sound four low beeps forever (no PPM). Of course i got //#define DUAL_INPUTS commemted and using the UART for debug

villamany, 2021-12-14:
However if i enable DEBUG_SERIAL_USART i should be able to send commands is correct?

I can read data like the calibration result, but if i send a command don't get any reply. IE i send HELP and no any reply

Candas1, 2021-12-14:
No it's not enough for sending commands, this should be also enabled.
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Debug-Serial#debug-protocol

villamany, 2021-12-14:
Sorry, i mean enable  DEBUG_SERIAL_PROTOCOL and no get any reply

Candas1, 2021-12-14:
Yes then it should work.
It could be both your green wires are not working.

villamany, 2021-12-14:
is there any way for using as PPM input the blue wire instead the green one?

villamany, 2021-12-14:
Modified this line in the defines.h files and got PPM train fine in Right side-blue wire.

#define PPM_PIN             GPIO_PIN_10 //GPIO_PIN_11

Candas1, 2021-12-14:
👍👍👍

Candas1, 2021-12-14:
Please check USART variant in the wiki.
I saw [this](https://github.com/nabeelfkhan/Toasty) by chance recently

villamany, 2021-12-17:
Hi, after some tests, R23 and R26 are faulty, some resistance but much more than 100ohm, so its a faulty board. I think probably some time ago in early tests i used the red wire of the port (15v) by mistake instead the 3v3 and fried resistors, with the  pots i get reads but very noisily.

Thanks for all the help!
[image]

closed 2021-12-17 by villamany