EFeru/hoverboard-firmware-hack-FOC #245: Hoverkart support

lolzika, 2021-12-14:
Hello there, i am looking to make a _long range_ hoverkart to tow my friends on a bike around.

long story short, i have a 10inch hoverboard and a hoverkart which operates by tilting the hoverboard. Since the active balancing and “normal” hoverboard use is not yet implemented i am wondering if i could use some version of the firmware in a way that the hoverboard gets strapped to the kart ( the levers are made immovable to not hinder operation under torque) and i use my rc radio (ibus compatible tx16s radio controller) together with a flysky receiver for the following:
- Regulate speed
- Turn left, right (with no servo but using the motors)
- Toggle between reverse, forward ans handbrake
- Toggle operation modes (high speed/high torque) i am not sure what is possible to change without modifying firmware onboard

Also will FOC give me better battery life (comparing to commutation at same speed) and higher possible speeds? Also will torque be affected by FOC?

If the above mentioned is possible i would love to know how powerfull is the integrated brake. (Can it slow me down to a stop on a 10% grade road) 
And i promise the last question lol, i assume this is a one way process (ie no coming back to the original FW as its locked). Does this firmware use the stock bootloader or does it have its own. Im thinking to remove the original mcu and just replacing it with a new one to try the FW

thanks for making this, it seems like a great project i cannot wait for hoverboard support to come out. (Cus when it does it is going on my daily driver board) 

Best regards,
Meltedgyro

EFeru, 2021-12-14:
Hi,

You can check the Wiki pages for what is currently possible with the firmware: https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki

For you I think you need to look at these VARIANTS:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-HOVERCAR
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-IBUS

I guess some of the questions will then be answered.
Regarding FOC, the motor control is better, also in terms of energy consumption and noise. The firmware has no bootloader needs "The bootloader is stored in the internal boot ROM (system memory) of STM32 devices, and is
programmed by ST during production." (taken from STM website)

Braking should be enough to stop a person. It is all about how much energy can be put back to the battery in a certain interval. If the battery is full, then less energy is going thought the batter, but the energy is there so the FETs can blow-up if you brake very aggresivelly. Personally, I did not experiment the board limits (I played the safe way... since I don't have many spare boards).

lolzika, 2021-12-14:
> 

Hi, thanks for the reply, using the hovercar variant, can i controll direction with the radio controller… is it possible to brake without pedals?

EFeru, 2021-12-14:
Yes, if you use RC control, then you have full control, directions, modes, etc

Jenya-m, 2021-12-14:
How can you make braking more powerful than traction? The fact is that the traction is quite strong and I have to limit it, but this reduces the braking force.

Jenya-m, 2021-12-14:
Now i am using:
#define I_MOT_MAX       25             // [A] Maximum single motor current limit
#define I_DC_MAX        27              // [A] Maximum stage2 DC Link current limit for Commutation and Sinusoidal types (This is the final current protection. Above this value, current chopping is applied. To avoid this make sure that I_DC_MAX = I_MOT_MAX + 2A)

I_DC_MAX = I_MOT_MAX + 2A -----Why is that?
----What if :
#define I_MOT_MAX       19            
#define I_DC_MAX         27   
----???