EFeru/hoverboard-firmware-hack-FOC #246: Different hall wiring?

villamany, 2021-12-17:
Hello, my motors some times dont start runing fine, tested all combinations of wiring but the only way i got nice work is swaping hall GPIO V and W in "defines.h file.

#define LEFT_HALL_U_PIN GPIO_PIN_5//GPIO_PIN_5
#define LEFT_HALL_V_PIN GPIO_PIN_7//GPIO_PIN_6
#define LEFT_HALL_W_PIN GPIO_PIN_6//GPIO_PIN_7

#define LEFT_HALL_U_PORT GPIOB
#define LEFT_HALL_V_PORT GPIOB
#define LEFT_HALL_W_PORT GPIOB

#define RIGHT_HALL_U_PIN GPIO_PIN_10//GPIO_PIN_10
#define RIGHT_HALL_V_PIN GPIO_PIN_12//GPIO_PIN_11
#define RIGHT_HALL_W_PIN GPIO_PIN_11//GPIO_PIN_12

Candas1, 2021-12-19:
Hi,

Yes both phases and halls can be different on some boards.

Candas1, 2021-12-19:
Do you share a picture of your mainboard?

villamany, 2021-12-19:
[image]

villamany, 2021-12-19:
[image]

Candas1, 2022-01-01:
Ok this board doesn't look different.
I am closing this issue as you figured out a workaround already.
Wiki is mentioning about the possibility of swapping phases/hall wires in case of weird behavior.

closed 2022-01-01 by Candas1