EFeru/hoverboard-firmware-hack-FOC #248: Can't get feedback on USART3 with control on USART3

lsisoft, 2021-12-27:
### Your System
- VARIANT: VARIANT_USART
- CONTROL TYPE: CONTROL_SERIAL_USART3
- CONTROL MODE: VLT_MODE
- What is your application?
    - experimenting with motor movement at this stage

Describe the bug and how we can reproduce it.

Part 1: I'm trying to utilize CONTROL_SERIAL_USART3 together with FEEDBACK_SERIAL_USART3 and hoverserial.ino to send motor speeds and get feedback back on the same serial, essentially RX (feedback) together with TX (speed command). Speed command works, the motor moves, but I don't get anything received back on USART3, essentially the arduino Receive() has always Serial.available() == 0.

I've decided to investigate this and hooked up a DEBUG_SERIAL_USART2 with a HC-06 bluetooth module and 9600 baud. Seen that plain printf-s send the data to serial and started to put printf log statements to confirm that feedback data is sent from the hoverboard. Essentially through the printf log I got confirmation that line: https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/cf7b1d0de1da34612b056c316480f47702a15cbe/Src/main.c#L479 is called. No idea why the data doesn't appear on Arduino.

Part 2: Tried to add DEBUG_SERIAL_PROTOCOL with DEBUG_SERIAL_USART2 and stopped receiving the debug serial messages. A possible conclusion is that the hoverboard uart implementation does either receive either send, but not both of them together.

I've put below my configuration that I'm playing with. The hoverserial.ino has minor change, to use lower speeds.

2 other things:
- I was pretty annoyed to get constant beep beeps while using hoverserial.ino. For a long while I though it is the serial timeout. While playing with DEBUG_SERIAL_USART2 I've peppered the code with printfs and found out that it was the backward motor beeping, like for a car. I highly recommend printfs to figure what the code does, when beeps are not clear enough.
- I've had lots and lots of problems when connecting STLink to the hoverboard (connection error, connection timeout, etc). In the end though a youtube video https://www.youtube.com/watch?v=XWh8yJ_p0HA (check position 01:00, 1 minute in the video) I found that STLInk clone don't really write the real pins on the outside plastic, but they are correct on the board underneath. Please consider adding this to the flashing page in your wiki.

```c
// ############################ VARIANT_USART SETTINGS ############################
#ifdef VARIANT_USART
  // #define SIDEBOARD_SERIAL_USART2 0
  //#define CONTROL_SERIAL_USART2  0    // left sensor board cable, disable if ADC or PPM is used! For Arduino control check the hoverSerial.ino
  //#define FEEDBACK_SERIAL_USART2      // left sensor board cable, disable if ADC or PPM is used!

  #define DEBUG_SERIAL_USART2           // left sensor cable debug
  #define USART2_BAUD           9600                  // UART2 baud rate (long wired cable)
  //#define DEBUG_SERIAL_PROTOCOL

  // #define SIDEBOARD_SERIAL_USART3 0
   #define CONTROL_SERIAL_USART3  0    // right sensor board cable. Number indicates priority for dual-input. Disable if I2C (nunchuk or lcd) is used! For Arduino control check the hoverSerial.ino
  #define FEEDBACK_SERIAL_USART3      // right sensor board cable, disable if I2C (nunchuk or lcd) is used!
  #define USART3_BAUD             115200
```

Candas1, 2021-12-27:
Hi,

Have you read [this](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Variant-USART) ?
Can be your the mainboard TX is defective on USART3, you can try USART2 instead.
[Backward beeps](https://github.com/EFeru/hoverboard-firmware-hack-FOC/blob/cf7b1d0de1da34612b056c316480f47702a15cbe/Inc/config.h#L177) can be disabled.

lsisoft, 2021-12-29:
Hi,

Thanks for the quick response.

I have read the 2 pages you mentioned.

Backward beeps can indeed be disabled, once you know they are backward beeps. Figuring out that they are backward beeps and not troubleshooting beeps was an 1h search, because hoverserial.ino managed to generate about 3-3 and a half beeps so made me thing is Serial Timeout. Could you add a comment line in hoverserial.ino to tell people to expect backward beeps?

Should I make a PR with these couple of ideas to merge into your main code?

I was able to get DEBUG_SERIAL_USART2 to work and provide feedback and I was able to send commands on CONTROL_SERIAL_USART3. USART2 was plugged into a bluetooth dongle HC-06 at 9600 bauds and I know for sure that dongle TX/RX work well.

I need to test if the Arduino RX on 0 pin works. I got the Arduino boards off AliExpress, maybe some pins are faulty. After htat will need to test the TX of USART3, on the hoverboard, if its working or not.

Candas1, 2021-12-29:
Usart variant/Arduino is one of the many use cases of the firmware.
Actually it's written [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Diagnostics)
I can definitely enhance the wiki if you have suggestions, it's far from perfect.

Candas1, 2021-12-29:
Yes this is how it's supposed to be

EFeru, 2022-01-01:
@lsisoft thanks for your PR #250  it is merged now. I will close this issue for now.

closed 2022-01-01 by EFeru

lsisoft, 2022-01-02:
@EFeru I've seen you removed from the PR, from README.md, the lines related to printf statements which work well on debug. Any particular reason?

Candas1, 2022-01-02:
https://github.com/EFeru/hoverboard-firmware-hack-FOC/wiki/Debug-Serial

It's in the wiki, we are going to simplify to the readme and remove all the information related to the HOW, all this should be in the wiki.